horizon-eda/horizon #237: Feature request: Longes pin names in symbol preview

fruchti, 2019-03-13:
It would be nice to be able to set some options available in the schematic editor in the symbol preview in the symbol editor, too. In particular,

I'm thinking of the “Pin display” setting: For all settings except “all”, the preview would automatically show the longest possible label for each pin. This would make testing whether all combinations of selected alternate pin names can fit inside the symbol a lot easier.

carrotIndustries, 2019-03-24:
This should be solved by https://github.com/carrotIndustries/horizon/commit/eba853ec89bef60b4bfc477e13f4735a9ca131d7 as that commit adds an option to expand symbols sideways in the schematic editor.

fruchti, 2019-03-25:
Great solution, thanks! I guess I should change the current wording in the convention

> 2. Make sure that pin names do not collide. Consider all alternate pin names.

so the default width does only include the default pin names?

carrotIndustries, 2019-03-26:
> so the default width does only include the default pin names?

Yes, and add a recommendation to set the "Can expand" option for symbols of units that have alternate names or where custom names make sense.

fruchti, 2019-03-26:
> > so the default width does only include the default pin names?
> 
> Yes, and add a recommendation to set the "Can expand" option for symbols of units that have alternate names or where custom names make sense.

Added it in 9c0afd1291943632b18a0e65374a887406a973d5

closed 2019-03-26 by fruchti