horizon-eda/horizon #240: Thoughts on the import DXF tool

atoav, 2019-03-28:
I use the `Import DXF` Tool in conjunction with the `line loop to polygon` tool to create panels, that I designed in Illustrator (I documented a working workflow [in this blog post](https://modular.atoav.com/blog/create-a-panel-using-illustrator-and-horizon-eda)).

There are three friction points:
## 1. The importer ignores all non-straight (e.g. bezier, arc) lines without warning
I fix this in Illustrator by subdividing round shapes first, and then simplyfing them to straight lines with a angle theshold (points on a line below that angle are removed). The most desireable thing would be if horizon would do just that on import. At least warning about this issue and suggesting a solution might be good here (took me quite a while to figure that one out).

## 2.  Line loop to polygon tool fails with shapes that have a donut-style shape
In the letters O, A, P, B, R, Q, D etc you have a "island" in the middle. The tool can't handle these shapes. My solution is to manually cut them into two halfs in Illustrator then export two dxf files where the shapes don't touch and use the line loop to polygon tool on them seperately and move them on top of each other on sucess. In illustrator it would look like this:
[image]  
Here the red and blue parts are exported and imported seperatly.  

I am not too familiar with the math behind this stuff, but I imagine the hardest pard is recognizing that kind of shapes. Technically I guess this is figuring out if one shape lies within another and then finding a way to slice them into as few independent parts as possible. 

## 3.  Line loop tool has to be called repeatedly
Because the tool only runs on the _first_ line loop found, I have to run it a hundred times by mashing <kbd>Space</kbd>, <kbd>Enter</kbd> in rapid succession. It would really help if there was a "All line loops to polygons" command, that just runs on all line loops found in the selection.

<hr>

Practically this is more or less a solved issue for me. But I am a power user of professional vector graphics software, so _I am able_ to solve problems like these (although with a ton on manual labour) – however for a beginner that tries to just import some panel, this might easily lead to a lot of frustration. I understand if you don't priorize this, I just wanted to note it here, while I still think about it : )

carrotIndustries, 2019-03-29:
> 1. The importer ignores all non-straight (e.g. bezier, arc) lines without warning

It's a bit more nuanced than that. Actual arcs are imported just fine. Converting beziers to straight line segments doesn't seem worth the effort (need to read up on the maths)

> At least warning about this issue and suggesting a solution might be good here (took me quite a while to figure that one out).

How to present this warning to the user? Display it in the tool bar at the bottom of the canvas?

> 2. Line loop to polygon tool fails with shapes that have a donut-style shape

This is a limitation of how polygons are represented in horizon. In horizon, polygons are stored as a contour consisting of line segments and arcs, so there's no way to store polygons with holes.

Adding holes to regular polygons would require quite significant changes to polygon-related functionality. Splitting up the polygon into simple polygons that don't need holes seems to be a more economic solution. If someone can point me to a suitable implementation, I could implement this in a tool.

> 3. Line loop tool has to be called repeatedly

Should be fixed in 0d97fd0fc1df6ba96f6ad257d05da3c7433db219

atoav, 2019-03-29:
> Converting beziers to straight line segments doesn't seem worth the effort (need to read up on the maths)  

I will have a look into this as well, I did this a while ago while I was working on a primitives library for the pygame engine. 

> How to present this warning to the user? Display it in the tool bar at the bottom of the canvas?  

Why not, maybe in red or something. A warning is better than nothing at all : )

>  In horizon, polygons are stored as a contour consisting of line segments and arcs, so there's no way to store polygons with holes.  

This is exactly what I thought it is that is, why my idea was to cut one `O` into two halves that touch`( )`. That is my manual fix for this. It works somehow like this:  
[image]  

Not sure if there exists a general solution for every (manifold) geometry though.  

> Should be fixed in 0d97fd0  
Thanks a lot, this will save me a ton of time, especially when I have to go back and forth : )

carrotIndustries, 2019-03-30:
> Why not, maybe in red or something. A warning is better than nothing at all : )

done in 21934d24eac3d70616da26d34a60d7cddd0e9f5f

atoav, 2019-04-05:
I won't close this for now for future reference, if it is ok for you