horizon-eda/horizon #250: First startup Experience: Pool Download

atoav, 2019-04-22:
I am in the process of checking over the "Getting Started" part of the docs, and I though I propose some changes.  

## Currently it works like this
1. Download the build
2. Run the `./horizon-eda.exe`
3. Click `Download pool...`
4. Select a destination
5. Go to Preferences and "activate" the pool

## Suggestion
My suggestion is: when there are no pools present on startup pop up a small window, with three choices:
- Download and activate the official pool
- Use an existing pool
- Create your own pool

This way we could skip 5 steps for most first time users _and_ it would be more comfortable for long time users who happen to install horizon on a new machine too. If we are extra smug we could consider adding a "what is a pool?" link to the documentation.

carrotIndustries, 2019-04-22:
> Select a destination

aaf1a65a24b780cfeb883d838321aca767258ae6 preselects a path in the users Documents directory

> Go to Preferences and "activate" the pool

That's done automatically for some time now.

>    Download and activate the official pool
>    Use an existing pool
>    Create your own pool

For most users, the third option likely isn't what they want, so I don't think offering that would be a sensible thing to do.

The second option should also mention that it's also supported to use git to manage the pool. 

I have something akin to https://docs.microsoft.com/en-us/windows/desktop/uxguide/ctrl-command-links in mind for the design of that dialog.

atoav, 2019-04-22:
> For most users, the third option likely isn't what they want, so I don't think offering that would be a sensible thing to do.

Okay you convinced me : )

What if we use the design you have in mind and add [this](https://user-images.githubusercontent.com/6620266/43834013-f633f792-9b0c-11e8-9f86-768b190cc319.jpg) to the left side of it? Makes it look more like a "Welcome, here are your options" thing, and less like nagging window you want to click away as fast as humanly possible.

carrotIndustries, 2019-04-24:
See https://github.com/carrotIndustries/horizon/tree/welcome_window for my take on that idea.
[image]

The copywriting and sizing of that window still needs some work

atoav, 2019-04-25:
Hey, not bad! I have no objections from a pure design side, the proportions are alright.

On windows it starts in a "expanded" width, probably it is about the text forcing the width?

[image]

The text for the "Download pool using pool manager" button needs some rework (as you said), because from a first time user perspective there is a lot of stuff to unpack here. I think we would do the user (and ourselves) a favour by framing the term "pool" a little different right from the start. There are two aspects to this:
- first time users might not be familiar with our meaning of "pool", so we need to give at least a hint
- first time users don't really care about *how* this download happens (pool manager), the want to know what it *means for them*

Something like: "Download community managed parts pool" would already help, because the "community managed" sounds friendly and tells them they can contribute if they want to and the "parts pool" makes clear what a pool could roughly be.

About the top text "Seems like this is the first time ...": it is alright, but I am conflicted whether it would be a good idea to give a hint about:
- the documentation
- where to report issues

Maybe this would be too much information yet again..

carrotIndustries, 2019-04-25:
> the documentation

definitely makes sense

> where to report issues

I guess that's better off in some sort of about dialog.

I also tried to include some brief explanation on what the pool is about in that window, but there's still scope for improvement.

[image]

atoav, 2019-04-26:
Okay this is much more informative now! Maybe even too much to read at the moment.

What I don't like about the top text: it tells the user they need to take the first option and then we go on to offer them another. It would be better to stay neutral and try to just inform there. On the other hand we can go a bit more emotional on the community part while staying with the truth:

[image]

**What I changed:**
- Split the top text into two paragraphs, one welcomes and explains, the next asks the question they have to answer
- Rephrase the text for the community pool button
- Add the line with the documentation link on the very bottom
- moved the whole buttons-and-text-block a little bit up to bring back visual balance that shifted towards the bottom once I added the footer line mentioning the documentation

I am still not too secure about the phrasing on the Download button explaination, what do you think?

carrotIndustries, 2019-04-28:
Maybe we also need to reword the second option:

Non first time users (users installing horizon on a second computer or such) will know what to to, dismiss the window and go straight to the preferences window.

What the second option should tell: Clone the pool yourself and then point horizon to it in the preferences window. Perhaps a git icon would be more appropriate?

atoav, 2019-04-28:
Mhh maybe there is a better way. We have Users who:
1. just wanna get started or check things out (=clone the community pool)
2. have a local pool "lying around" (=select some path)
3. have a different pool in a remote repository (=clone their own pool)

I know you implied step 3 was already taken for those who take step 2, but what if the second option would offer both? Select a local path or enter a remote repo. I don't really mind having to run a `git clone https://github.com/atoav/my-friggin-pool` for _once_ before beeing able to select a pool, so I think we could just skip that.

For me the folder icon is quite clear: open something from disk. Whether it is managed in git or not doesn't really matter to be honest. I would change it from the future tense to the present tense. So instead of speaking what users plan to do, I would say somthing like: "Use this option if you already downloaded a pool or manage your own."

carrotIndustries, 2019-04-29:
> have a different pool in a remote repository (=clone their own pool)

I wanted to make clear that folks who already know git might be better off using git instead of the pool manager's git integration that tries to hide the git repository from the user as good as possible. That's what I tried to convey in the second option.

Since that window is primarily aimed at first-time users, maybe we should make the second button have a git icon instead and tell the users that they need to clone the pool somewhere and point to pool manager to it.

Opening an existing pool could then be a smaller button at the bottom of the window.

atoav, 2019-04-30:
Ah okay, but the switch (at least in my case was quite seamless. I started using the project manager and then just switched to using git in that same pool directory manually (adding my fork as a remote).

[image]

Maybe then it would be clever to use three options:
1. **Download community managed pool** (Icon: "↓")
Update your copy of the pool with the community's contributions or take part yourself
2. **Use your own repository** (Icon: "Git")
Manage your own pool or a fork of the community pool using git yourself (for experts)
3. **Use a existing pool** (Icon: "Folder")
Open a existing pool from disk and activate it

I am a little worried however that this leads to a little bit of confusion:
- When you select option 2, but you don't already have a fork or a pool you managed yourself, wouldn't your expectation be, that you could _create a pool_ or _get a fork_ using this option? And when you already have a repo your expectations would be to enter username and repo name and get it cloned and selected in the preferences in one go
- The difference between options 2 and 3 is a bit vague, because technically you could select option 3 and still manage the pool with git, but I think this is a minor issue and can be ignored.

What do you think? Do 3 options make more sense?

carrotIndustries, 2019-05-04:
[image]

Clicking the git button will show this page:

[image]

Users who already know git will likely be more comfortable doing these steps themselves than having the pool manager do it for them.

Technically, it's the same as clicking on the "Use existing pool" button.

atoav, 2019-05-06:
[image]

I tried to give it a spin, but to my surprise, I didn't get the welcome window but this. Is this intentional?
Details: I renamed my horizon folder at `%APPDATA` to `horizon_OLD` so it doesn't get picked up

atoav, 2019-05-06:
Okay very cool, I think this is the way to go. A few details:
- When you go with Option 1 and download the community pool, it opens the pool manager afterwards. Given our understanding that this might be the "beginner option", wouldn't it be less confusing to show the main window instead?
- I could imagine that it would make sense to add an Documentation page called "How to manage your own pool via git" and then include that link in the git sub-page. As a examplary workflow on things like how you keep your pool updated with the upstream, how to use branches for new parts etc. But maybe we should include this once the Documentation exists.

carrotIndustries, 2019-05-06:
> Given our understanding that this might be the "beginner option", wouldn't it be less confusing to show the main window instead?

added in 858563bb656509a9d5056009475f0c1aef33b3b6.

While we're at it, how about adding some secondary kind of  "What next" window that pops up after the download has completed?

That window doesn't need to contain much content, I had the idea of including links to existing projects so that users get an idea of what's possible and links to the getting started docs.

> But maybe we should include this once the Documentation exists.

Yes, since that's only changing the text a bit.

carrotIndustries, 2019-05-12:
Since the other first run windows are more or less independent, the only thing left to do is to finalize the artwork on the left.

atoav, 2019-05-12:
Okay discussion about that can go on in #192

One thing: one of my hard drives broke so the source files for that splash specifically are lost because I didn't have them in the backup for some reason. I could redo another one, or we could use that one, however you like. Redoing another one could make sense if we want higher resolution, but I am not sure how close I'd get to the same thing again.

carrotIndustries, 2019-08-27:
If no one objects, I'll merge this in the next days using the artwork from https://github.com/horizon-eda/horizon-artwork/blob/master/splash/PNG/splash1_1080.png

atoav, 2019-08-27:
No objections.

carrotIndustries, 2019-08-27:
finally merged in 877a53300a2628dad1b35a0e7357502a6f999e5b

closed 2019-08-27 by carrotIndustries