horizon-eda/horizon #251: [UX] Minor usage inconsistency between Draw Arc and Draw Polygon

atoav, 2019-04-23:
I am in the process of writing the documentation section on the drawing tools and found a minor usability inconsistency between the "Draw arc" and the "Draw polygon" Actions:  

If you draw a line arc you press <kbd>d a</kbd>, set the arc start, the arc end, can then flip the arc direction by pressing <kbd>e</kbd> and finally set the arcs center.

If you draw a polygon you press <kbd>d y</kbd>, make the edge an arc by pressing <kbd>a</kbd> and then you set the arc center, flip the arc by pressing <kbd>e</kbd> and then the end point.

So the line arc is `Start → End→ (Flip) → Center` and the polygon is `Start → Center → (Flip) → End`, which caused even myself as a long time user quite some confusion when drawing polygon arcs, I recount spending 5 minutes to draw a polygon circle once, because I didn't had the concentration to lean back and think systematically about it (now we have gladly have the draw polygon circle Action).

## Solution
The order used by the line arc tool feels more natural to me and is more intuitive, when it comes to changing the arc direction. I suspect the reason for the arc used in the polygon tool has to do with the underlying data structure?

I'd highly recommend changing every arc to work like the Draw line arc Action. If this is not possible for the polygon tool, because of data structure reasons, we could still make the Draw line Arc tool work the same as the polygon arc tool..

carrotIndustries, 2019-04-26:
Have a look at https://github.com/carrotIndustries/horizon/tree/draw-polygon-arc

atoav, 2019-04-27:
Works pretty well, I have no objections to merge that one from a usability perspective, I will update the docs too. So I think I can close this one?

closed 2019-04-27 by atoav