horizon-eda/horizon #252: Expand default keybindings

atoav, 2019-04-23:
It might be a good idea to bind more shortcuts for the release. We don't have to bind everything, but:
- things where it makes sense from a consistency standpoint (e.g. binding "Rotate arbitrary" to <kbd>R</kbd> just like with "move" <kbd>m</kbd> and "Move exactly <kbd>M</kbd>)
- things that might be used heavily (e.g. "Place on Board" in the Schematic Editor)

You get the idea. Should I draft something up or should I just fork the thing? Where are the default keybindings defined?

carrotIndustries, 2019-04-23:
Great idea, the default keybindings are defined in https://github.com/carrotIndustries/horizon/blob/master/src/imp/keys_default.json get baked into the binary during the build.

The easiest way to add keybindings is to set them in the app itself, close the preferences dialog and copy the relevant object (a child of `key_sequences`) from the `prefs.json` in your appdata (local) directory to `keys_default.json`.

Unfortunately, there's a bit of an issue with the installed base: The default key bindings only get applied if there isn't an entry (even an empty one) in the user's `prefs.json` for the tool in question. The reasoning for this behaviour was to make it possible for users to delete keybindings. (Otherwise they'd reappear after reopening the application). To put it short: We currently have no way of knowing whether the user has explicitly deleted the default key sequences or there just haven't ever been any.

One way out of this misery would be to track if the user has modified the keybindings and apply defaults if the user hasn't touched the keybindings.

atoav, 2019-04-26:
Did you have any kind of system or philosophy when choosing the shortcuts? I won't change existing ones, but keeping some sort of consistency could make sense. E.g. <kbd>p</kbd> seems to be predominantly used for "place", while <kbd>d</kbd> is used for draw etc. I noticed you used <kbd>i</kbd> for "Edit" once, maybe we can reuse that in other instances as well.

carrotIndustries, 2019-04-26:
> Did you have any kind of system or philosophy when choosing the shortcuts?

I'm afraid, not really. Trying to give rationale for some the non-obvious ones:
 - <kbd>i</kbd> for edit since `e` and `d` were already taken (choose first unused letter)
 - <kbd>e</kbd> for mirror since it's that way in LTSpice
 - <kbd>x</kbd> to start a track because kicad
 - <kbd>q</kbd> to update planes as `q` is after `p` in the alphabet
 - <kbd>k</kbd> to add a junction to a net line because `knick`
 - <kbd>/</kbd> for placing a bus ripper since these look a bit like a `/`

f00e9c9ef5cab6d3c91e1a9208bdf37cbc57c386 allows us to use <kbd>i</kbd> for edit consistently for all object types

Having thought about whether to store the 'key bindings modified by user' flag, I think that's the right thing to do, so we can add key bindings to existing actions.

For the installed base, just marking all key bindings as default non modified and thus deleting users' custom key bindings seems to be an acceptable trade-off.

carrotIndustries, 2019-04-30:
More or less forget what I said regarding the issues regarding existing key bindings:
The issue was that clicking on an action in the keys preferences (not adding a key sequence) created the entry for that action resulting in new default key sequences not being applied. 729a7af735d98a919c26691e1251598f215c4a00 fixes this.

Users might still not get the new defaults automatically due to this. But since we now have that "Load Default" button that shouldn't be that big an issue.

carrotIndustries, 2019-11-24:
Closing since f152e2a4ab2d25d2c1c3935a1982841b8685bbc2 adds more keybindings where useful.

closed 2019-11-24 by carrotIndustries