horizon-eda/horizon #254: Compilation issue on openSUSE

niclashoyer, 2019-05-07:
On openSUSE Tumbleweed I get the following message:

```bash
g++ src/util/uuid.o src/util/uuid_path.o src/util/sqlite.o src/util/str_util.o src/pool/pool_manager.o src/pool/unit.o src/pool/symbol.o src/pool/part.o src/common/common.o src/common/junction.o src/common/line.o src/common/arc.o src/common/layer_provider.o src/pool/gate.o src/block/net.o src/block/bus.o src/block/block.o src/pool/entity.o src/block/component.o src/block/bom.o src/block/bom_export_settings.o src/schematic/schematic.o src/schematic/sheet.o src/common/text.o src/schematic/line_net.o src/schematic/net_label.o src/schematic/bus_label.o src/schematic/bus_ripper.o src/schematic/schematic_symbol.o src/schematic/power_symbol.o src/schematic/schematic_rules.o src/schematic/rule_single_pin_net.o src/schematic/pdf_export_settings.o src/pool/padstack.o src/common/polygon.o src/common/hole.o src/common/shape.o src/common/patch_type_names.o src/pool/package.o src/package/pad.o src/package/package_rules.o src/package/rule_package_checks.o src/board/board.o src/board/board_package.o src/board/track.o src/board/via_padstack_provider.o src/board/via.o src/board/plane.o src/board/board_rules.o src/board/rule_hole_size.o src/board/rule_clearance_silk_exp_copper.o src/board/rule_track_width.o src/board/rule_clearance_copper.o src/board/rule_parameters.o src/board/rule_via.o src/board/rule_clearance_copper_other.o src/board/rule_plane.o src/board/rule_diffpair.o src/board/rule_preflight_checks.o src/board/rule_clearance_copper_keepout.o src/board/airwires.o src/board/fab_output_settings.o src/board/board_hole.o src/board/connection_line.o src/pool/pool.o src/util/placement.o src/util/util.o src/util/csv.o src/common/object_descr.o src/block/net_class.o src/project/project.o src/resources.o src/gitversion.o src/rules/rules.o src/rules/rule.o src/rules/rule_descr.o src/rules/rule_match.o src/rules/rule_match_keepout.o src/parameter/program.o src/parameter/set.o 3rd_party/clipper/clipper.o src/common/dimension.o src/logger/logger.o src/parameter/program_polygon.o src/frame/frame.o src/common/keepout.o src/board/board_layers.o src/pool/pool_parametric.o 3rd_party/router/router/pns_router.o 3rd_party/router/router/pns_item.o 3rd_party/router/router/pns_node.o 3rd_party/router/router/pns_solid.o 3rd_party/router/router/pns_optimizer.o 3rd_party/router/router/pns_topology.o 3rd_party/router/router/pns_walkaround.o 3rd_party/router/router/pns_utils.o 3rd_party/router/router/pns_algo_base.o 3rd_party/router/router/pns_diff_pair_placer.o 3rd_party/router/router/pns_diff_pair.o 3rd_party/router/router/pns_dp_meander_placer.o 3rd_party/router/router/pns_dragger.o 3rd_party/router/router/pns_itemset.o 3rd_party/router/router/pns_line_placer.o 3rd_party/router/router/pns_line.o 3rd_party/router/router/pns_via.o 3rd_party/router/router/pns_logger.o 3rd_party/router/router/pns_meander_placer_base.o 3rd_party/router/router/pns_meander_placer.o 3rd_party/router/router/pns_meander_skew_placer.o 3rd_party/router/router/pns_meander.o 3rd_party/router/router/pns_shove.o 3rd_party/router/router/time_limit.o 3rd_party/router/router/pns_routing_settings.o 3rd_party/router/router/pns_sizes_settings.o 3rd_party/router/common/geometry/shape_line_chain.o 3rd_party/router/common/geometry/shape_arc.o 3rd_party/router/common/geometry/shape.o 3rd_party/router/common/geometry/shape_collisions.o 3rd_party/router/common/geometry/seg.o 3rd_party/router/common/geometry/geometry_utils.o 3rd_party/router/common/math/math_util.o 3rd_party/router/wx_compat.o src/router/pns_horizon_iface.o src/core/tool_route_track_interactive.o src/util/step_importer.o src/export_step/export_step.o src/imp/imp_main.o src/imp/main_window.o src/imp/imp.o src/imp/imp_layer.o src/imp/imp_symbol.o src/imp/imp_schematic.o src/imp/imp_padstack.o src/imp/imp_package.o src/imp/imp_board.o src/imp/imp_frame.o src/imp/tool_popover.o src/imp/selection_filter_dialog.o src/imp/action.o src/imp/action_catalog.o src/canvas/canvas.o src/canvas/canvas_gl.o src/canvas/canvas_cairo.o src/canvas/grid.o src/canvas/gl_util.o src/canvas/pan.o src/canvas/render.o src/canvas/draw.o src/canvas/text.o src/canvas/hershey_fonts.o src/canvas/drag_selection.o src/canvas/selectables.o src/canvas/hover_prelight.o src/canvas/triangle.o src/canvas/image.o src/canvas/selection_filter.o 3rd_party/polypartition/polypartition.o src/canvas/marker.o src/canvas/annotation.o src/canvas/fragment_cache.o src/util/text_data.o 3rd_party/poly2tri/common/shapes.o 3rd_party/poly2tri/sweep/cdt.o 3rd_party/poly2tri/sweep/sweep.o 3rd_party/poly2tri/sweep/sweep_context.o 3rd_party/poly2tri/sweep/advancing_front.o src/util/msd.o src/util/msd_animator.o src/canvas/appearance.o src/core/core.o src/core/core_properties.o src/core/tool_move.o src/core/tool_place_junction.o src/core/tool_draw_line.o src/core/tool_delete.o src/core/tool_draw_arc.o src/core/tool_map_pin.o src/core/tool_map_symbol.o src/core/tool_draw_line_net.o src/core/tool_place_text.o src/core/tool_place_net_label.o src/core/tool_disconnect.o src/core/tool_bend_line_net.o src/core/tool_move_net_segment.o src/core/tool_place_power_symbol.o src/core/tool_edit_symbol_pin_names.o src/core/tool_place_bus_label.o src/core/tool_place_bus_ripper.o src/core/tool_manage_buses.o src/core/tool_draw_polygon.o src/core/tool_enter_datum.o src/core/tool_place_hole.o src/core/tool_place_pad.o src/core/tool_paste.o src/core/tool_assign_part.o src/core/tool_map_package.o src/core/tool_draw_track.o src/core/tool_place_via.o src/core/tool_route_track.o src/core/tool_drag_keep_slope.o src/core/tool_add_part.o src/core/tool_smash.o src/core/tool_place_shape.o src/core/tool_edit_shape.o src/core/tool_import_dxf.o src/core/tool_edit_parameter_program.o src/core/tool_edit_pad_parameter_set.o src/core/tool_draw_polygon_rectangle.o src/core/tool_draw_line_rectangle.o src/core/tool_edit_line_rectangle.o src/core/tool_helper_map_symbol.o src/core/tool_helper_move.o src/core/tool_helper_merge.o src/core/tool_edit_via.o src/core/tool_rotate_arbitrary.o src/core/tool_edit_plane.o src/core/tool_update_all_planes.o src/core/tool_draw_dimension.o src/core/tool_set_diffpair.o src/core/tool_select_more.o src/core/tool_set_via_net.o src/core/tool_lock.o src/core/tool_add_vertex.o src/core/tool_place_board_hole.o src/core/tool_edit_board_hole.o src/core/tool_generate_courtyard.o src/core/tool_set_group.o src/core/tool_copy_placement.o src/core/tool_copy_tracks.o src/core/tool_swap_nets.o src/core/tool_line_loop_to_polygon.o src/core/tool_change_unit.o src/core/tool_helper_line_width_setting.o src/core/tool_set_nc_all.o src/core/tool_set_nc.o src/core/tool_add_keepout.o src/core/tool_helper_draw_net_setting.o src/core/tool_helper_get_symbol.o src/core/tool_change_symbol.o src/core/tool_place_refdes_and_value.o src/core/tool_helper_restrict.o src/core/tool_draw_polygon_circle.o src/core/tool_draw_connection_line.o src/core/tool_backannotate_connection_lines.o src/core/cores.o src/core/clipboard.o src/core/buffer.o src/dialogs/map_pin.o src/dialogs/map_symbol.o src/dialogs/map_package.o src/dialogs/ask_net_merge.o src/dialogs/ask_delete_component.o src/dialogs/select_net.o src/dialogs/symbol_pin_names.o src/dialogs/manage_buses.o src/dialogs/ask_datum.o src/dialogs/ask_datum_string.o src/dialogs/select_via_padstack.o src/dialogs/dialogs.o src/dialogs/annotate.o src/dialogs/edit_shape.o src/dialogs/manage_net_classes.o src/dialogs/manage_power_nets.o src/dialogs/edit_parameter_program.o src/dialogs/edit_parameter_set.o src/dialogs/edit_pad_parameter_set.o src/dialogs/edit_via.o src/dialogs/pool_browser_dialog.o src/dialogs/schematic_properties.o src/dialogs/edit_plane.o src/dialogs/edit_stackup.o src/dialogs/edit_board_hole.o src/dialogs/edit_frame.o src/dialogs/edit_keepout.o src/dialogs/select_group_tag.o src/dialogs/ask_datum_angle.o src/util/sort_controller.o src/core/core_symbol.o src/core/core_schematic.o src/core/core_padstack.o src/core/core_package.o src/core/core_board.o src/core/core_frame.o src/property_panels/property_panels.o src/property_panels/property_panel.o src/property_panels/property_editor.o src/widgets/warnings_box.o src/widgets/net_selector.o src/widgets/sheet_box.o src/widgets/net_button.o src/widgets/layer_box.o src/widgets/spin_button_dim.o src/widgets/chooser_buttons.o src/widgets/cell_renderer_color_box.o src/widgets/net_class_button.o src/widgets/parameter_set_editor.o src/widgets/pool_browser.o src/widgets/component_selector.o src/widgets/compone [... cut ...]
collect2: fatal error: cannot find 'ld'
compilation terminated.
make: *** [Makefile:568: horizon-imp] Error 1
```

But `ld` is in `$PATH`:

```bash
$ which ld
/usr/bin/ld
$ ld --version
GNU ld (GNU Binutils; openSUSE Tumbleweed) 2.32
Copyright (C) 2019 Free Software Foundation, Inc.
This program is free software; you may redistribute it under the terms of
the GNU General Public License version 3 or (at your option) a later version.
This program has absolutely no warranty.
```

Any suggestions? Am I missing a package?

carrotIndustries, 2019-05-07:
The makefile tells gcc to use the gold linker:
```
-fuse-ld=gold
```
Perhaps that's an extra package on suse? You may also try manually removing this flag.

niclashoyer, 2019-05-07:
Thanks that worked, I added the dependency to the PR: https://github.com/carrotIndustries/horizon-docs/pull/8

Are these warnings from the linker critical?

```bash
/usr/bin/ld.gold: warning: skipping incompatible /lib/libgcc_s.so.1 while searching for libgcc_s.so.1
/usr/bin/ld.gold: warning: skipping incompatible /lib/libgcc_s.so.1 while searching for libgcc_s.so.1
```

carrotIndustries, 2019-05-07:
> Are these warnings from the linker critical?

Dunno. I guess not if the resulting binaries work.

niclashoyer, 2019-05-07:
Ok, I'm testing right now, so far it looks good! :+1:

closed 2019-05-07 by niclashoyer