horizon-eda/horizon #255: Feature Request: Mutliple / stacked Boards

niclashoyer, 2019-05-07:
I'm currently using KiCad fpr my projects and I have some projects that have a fixed width for the PCBs, so that they need to be stacked with pin headers.

The current workflow in KiCad is very cumbersome. I usually add two board files and import the netlist to each of them. Then deleting the components I don't want on that board until I have them separated. There are DRC errors then, because not all components are placed, but there is no real way around this.

I'm open for discussion on this. LibrePCB supports multiple boards, but it uses the same netlist for each, so that would be a different use case (e.g. one THT and one SMD board of the same schematic).

I think a kind of "sub board" feature (like sub schematic) would be useful, so that the board can be split up in several boards while the netlist could be checked for all boards. The gerber export then needs to be done for each "sub board".

carrotIndustries, 2019-05-12:
Unfortunately, I don't see this feature landing anytime soon due to architectural difficulties.

pwnorbitals, 2020-10-10:
I think this would be a very useful feature. Not sure how that would work in code, though

bob-u, 2020-10-10:
A half way feature could be ability to convert a PCB layout into a foot print.