horizon-eda/horizon #256: Feature Request: command line interface

niclashoyer, 2019-05-07:
Provide a command line interface to automate creation of BOM, gerber export or DRC.

I tried this with KiCad (which has no real CLI) and resorted to a docker container running xvfb which is a PITA and breaks continuously.

carrotIndustries, 2019-05-07:
I've though about this as well and it's certainly within the realms of possibility. One reason why we have no CLI yet is not having a clear idea of what it should be like:

 - Provide a limited set of features such as check drc, export pdf, etc accessible using command line args
 - Launch some sort of script interpreter that you'd tell to open the schematic and export the pdf (TCL would be the classical choice)
 - Create a python module (implemented in C/C++, calling into the horizon code) to the same as above:

From a programmer's perspective, having something based on python to achieve the latter would be preferable, but that doesn't package very well on windows.

Just writing from the top of my head of how that python thing could look like
```
>>> import horizon
>>> brd = horizon.load_board("path/to/project.hprj")
>>> settings = brd.get_fab_output_settings() #returns the fab output settings serialized to a dict
>>> #do something with the settings
>>> brd.export_gerber(settings)
```

Since windows users probably aren't interested in that kind of feature at all, the packaging issues are acceptable from my perspective.

niclashoyer, 2019-05-08:
If the python interface is simple to use, that would be the best idea. But if you take KiCad as an example, which also has a python interface for the board designer, it gets complicated quickly, even for [simple tasks as checking DRC](https://forum.kicad.info/t/kicad-drc-for-travis/13852/4). There are even blogs just about [kicad scripting](https://kicad.mmccoo.com/). There are a lot of python scripts out there for KiCad, but most of them are unmaintained, don't do what the actually should do, or reinvent the wheel. Of course one can do fancy stuff with that (e.g. generative board outlines), but for simple tasks this might be too much.

Maybe a limited CLI and a scripting interface is orthogonal, so it may be good to have both. From an CI perspective you'd just want to check the rules, convert schematic to pdf, export BOM, convert the board(s) to gerber/pdf.

carrotIndustries, 2019-05-08:
> If the python interface is simple to use, that would be the best idea.
> Maybe a limited CLI and a scripting interface is orthogonal

I'm not planning on exposing all of horizon's guts in a python module. That module will just provide features similar to the ones you mentioned.

carrotIndustries, 2019-05-12:
5bc72b48340893d4ca29f9870d24814cf1598f03 adds a python module that allows for exporting BOM and PDF

run `make horizon.so` to build the module

Usage:
```
import sys
sys.path.append(".")
import horizon
p=horizon.Project("/path/to/project.hprj")
sch = p.open_top_schematic()

pdf_settings = sch.get_pdf_export_settings()
pdf_settings['output_filename'] = '/tmp/sch.pdf'
sch.export_pdf(pdf_settings)

bom_settings = sch.get_bom_export_settings()
bom_settings['output_filename'] = '/tmp/bom.csv'
sch.export_bom(bom_settings)
```

carrotIndustries, 2020-07-25:
closing since the pyhon modules now also does DRC

closed 2020-07-25 by carrotIndustries