horizon-eda/horizon #257: Problem with Gerber export and polygon with rounded corners

niclashoyer, 2019-05-10:
I used the polygon tool (`d Y`) for the outline of my test board and added 2 mm rounded corners (using `r`). That went very well and the output also seems ok, looks good in the 3D viewer and also in Gerbv.

After uploading the gerbers to [AISLER](https://aisler.net), their milling algorithm crashed. OSHPark could render the output, but had some issues.

I got in touch with the support, here is the answer:

> Our milling algorithm crashes during the import. Reason for this is the boardoutline which consist of 260 single lines, which overlap in a strange way. If you just use a rectangular shape it should be fine.

Using a rectangular shape worked.

Does horizon generate the arcs for the rounded corners in a strange way? Gerber should support arcs natively, shouldn't it? So there shouldn't be any approximation using many lines around the corners.

carrotIndustries, 2019-05-10:
> Does horizon generate the arcs for the rounded corners in a strange way? Gerber should support arcs natively, shouldn't it? So there shouldn't be any approximation using many lines around the corners.

In order to get horizon where it is today, some shortcuts had to be taken, always replacing arcs by straight line segments was one of them as it made things easier.

closed 2019-05-10 by carrotIndustries

niclashoyer, 2019-05-10:
Thank you so much! I'm very impressed where horizon is today. I hope it really takes off. First board going to production soon :tada:

commit 1008f9dd (2019-05-10): use arcs in gerber outline output

fixes #257