horizon-eda/horizon #261: Cloning community pool doesn't work: HTTP 301 response

atoav, 2019-05-12:
I suspect this has to do with the recent move:  
[image]  

At the same time my pool fails to load up with the Error:  
[image]

I have to investigate further, to figure out what's going on

atoav, 2019-05-12:
I tried with a newer build 3cb41189, and have the same happening to me. Manually cloning and opening the pool works, but for some reason my own pool fails with that UUID error, still investigating...

carrotIndustries, 2019-05-12:
The first problem is solved by dfa9fad3d884b31c1a987e3d99dd3cf84acdd9a6 and db12787592138cde53a9ab3c567da0a93d20457d

However, I can't think of a reason for the second one...

atoav, 2019-05-12:
I tried copying the working `pool.json` from upstream/master to my pool, with no success. How can I trace down the origin of that invalid UUID?

carrotIndustries, 2019-05-12:
> How can I trace down the origin of that invalid UUID?

 - Compile horizon from source so that you have debug symbols
 - From the mingw shell, run `gdb ./horizon-eda`
 - In the gdb shell:
 - `catch throw`
- `run`
- Do the action that causes the error
- in the gdb shell, type `bt` to get the backtrace

You may also try deleting the `pool.db` and `parametric.db` file.

atoav, 2019-05-12:
Thanks, I solved it. I just cloned my pool freshly and set up the remote to the new URL. For some reason it now works. The remote adress shouldn't have anything to do with it, but the problem seems solved now : )

closed 2019-05-12 by atoav

atoav, 2019-05-12:
Okay also figured out the cause:

I was trying to work out a new `table.json` for polarized capacitors and this seems to have impacted the `parametric.db`. See the file here: https://gist.github.com/atoav/0abd403bdb419a85e964e240c261e309

atoav, 2019-05-12:
Okay very cool this works. Only a short hint:
[image]
  
Is there any way to format the hours and the temperature here without exponent?