horizon-eda/horizon #266: invalid UUID

frmdstryr, 2019-05-20:
I just pulled from master and built, now trying to open any project or the pool gives a UUID error. 

[image]

[image]

How can I see what UUID is the problem here?

frmdstryr, 2019-05-20:
The uuid is an empty string.

carrotIndustries, 2019-05-20:
launch horizon in gdb:
```
gdb ./horizon-eda
(gdb) catch throw
(gdb) run
```

When gdb breaks, type `bt` to show a backtrace. Then, `cont` to continue the program.

frmdstryr, 2019-05-20:
```
Thread 1 "horizon-eda" hit Catchpoint 1 (exception thrown), 0x00007ffff3680ced in __cxa_throw () from /usr/lib/x86_64-linux-gnu/libstdc++.so.6
(gdb) bt
#0  0x00007ffff3680ced in __cxa_throw () at /usr/lib/x86_64-linux-gnu/libstdc++.so.6
#1  0x000055555567adf6 in horizon::UUID::UUID(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&) ()
#2  0x00005555559a182b in horizon::PoolBrowserParametric::search() (this=0x555555dfc530) at /usr/include/c++/8/ext/new_allocator.h:86
#3  0x00005555559a4c70 in horizon::PoolBrowserParametric::PoolBrowserParametric(horizon::Pool*, horizon::PoolParametric*, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
    (this=0x555555dfc530, p=<optimized out>, pp=<optimized out>, table_name=..., __in_chrg=<optimized out>, __vtt_parm=<optimized out>)
    at src/widgets/pool_browser_parametric.cpp:234
#4  0x00005555558e916b in horizon::PoolNotebook::PoolNotebook(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&, horizon::PoolProjectManagerAppWindow*)
    (this=0x555556506520, bp=..., aw=<optimized out>, __in_chrg=<optimized out>, __vtt_parm=<optimized out>)
    at src/pool-prj-mgr/pool-mgr/pool_notebook.cpp:235
#5  0x00005555558da56f in horizon::PoolProjectManagerAppWindow::open_file_view(Glib::RefPtr<Gio::File> const&) (this=0x555555ffb000, file=...)
    at src/pool-prj-mgr/pool-prj-mgr-app_win.cpp:907
#6  0x00005555558db35c in horizon::PoolProjectManagerAppWindow::<lambda(Gtk::ListBoxRow*)>::operator()
    (__closure=<optimized out>, row=<optimized out>) at src/pool-prj-mgr/pool-prj-mgr-app_win.cpp:187
#7  0x00005555558db35c in sigc::adaptor_functor<horizon::PoolProjectManagerAppWindow::PoolProjectManagerAppWindow(Gtk::ApplicationWindow::BaseObjectType*, const Glib::RefPtr<Gtk::Builder>&, horizon::PoolProjectManagerApplication*)::<lambda(Gtk::ListBoxRow*)> >::operator()<Gtk::ListBoxRow* const&> (_A_arg1=<optimized out>, this=<optimized out>) at /usr/include/sigc++-2.0/sigc++/adaptors/adaptor_trait.h:89
#8  0x00005555558db35c in sigc::internal::slot_call1<horizon::PoolProjectManagerAppWindow::PoolProjectManagerAppWindow(Gtk::ApplicationWindow::BaseObjectType*, const Glib::RefPtr<Gtk::Builder>&, horizon::PoolProjectManagerApplication*)::<lambda(Gtk::ListBoxRow*)>, void, Gtk::ListBoxRow*>::call_it(sigc::internal::slot_rep *, sigc::type_trait_take_t) (rep=<optimized out>, a_1=<optimized out>)
    at /usr/include/sigc++-2.0/sigc++/functors/slot.h:148
#9  0x00007ffff7022813 in  () at /usr/lib/x86_64-linux-gnu/libgtkmm-3.0.so.1
#10 0x00007ffff479210d in g_closure_invoke () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#11 0x00007ffff47a4bf1 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#12 0x00007ffff47ad715 in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#13 0x00007ffff47ae12f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#14 0x00007ffff590c150 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#15 0x00007fffeeb98dae in ffi_call_unix64 () at /usr/lib/x86_64-linux-gnu/libffi.so.6
#16 0x00007fffeeb9871f in ffi_call () at /usr/lib/x86_64-linux-gnu/libffi.so.6
#17 0x00007ffff4792ced in g_cclosure_marshal_generic_va () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#18 0x00007ffff4792346 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#19 0x00007ffff47ad9ff in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#20 0x00007ffff47ae12f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#21 0x00007ffff58cba36 in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#22 0x00007ffff4795008 in g_cclosure_marshal_VOID__BOXEDv () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#23 0x00007ffff4792346 in  () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#24 0x00007ffff47ad9ff in g_signal_emit_valist () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#25 0x00007ffff47ae12f in g_signal_emit () at /usr/lib/x86_64-linux-gnu/libgobject-2.0.so.0
#26 0x00007ffff58c8d0e in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#27 0x00007ffff58ca2fb in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0
#28 0x00007ffff58ccf5e in  () at /usr/lib/x86_64-linux-gnu/libgtk-3.so.0

```

frmdstryr, 2019-05-20:
https://github.com/horizon-eda/horizon/blob/cf201660fb39cbd2afe2b156ea1dcbb36eadeb5b/src/widgets/pool_browser_parametric.cpp#L346

frmdstryr, 2019-05-20:
Sorry its this line https://github.com/horizon-eda/horizon/blob/master/src/widgets/pool_browser_parametric.cpp#L360

frmdstryr, 2019-05-20:
It's the cast from ` q.get<std::string>(ofs + 4)` to a UUID.  Not sure why the pool_uuid is empty?

carrotIndustries, 2019-05-20:
> and updating the pool is crashing.

Looks like we're onto something. Can you do a backtrace of that as well?

frmdstryr, 2019-05-20:
If it helps, I pushed my pool here: https://github.com/frmdstryr/horizon-pool/tree/current

carrotIndustries, 2019-05-20:
Hm, I still cannot reproduce this. However, I've had some issues with the padstacks lately as well, doing a `make clean` and building again from scratch fixed things.

frmdstryr, 2019-05-20:
That worked... sorry for the noise...

closed 2019-05-20 by frmdstryr