horizon-eda/horizon #268: Error: unexpected HTTP response code 307 when accessing https://api.github.com/repos/carrotIndustries/horizon-pool/pulls

krasimirk, 2019-05-20:
This error is shown when I create a new pull request from the pool manager:
[image]

However the pull request was visible in the github web site.

carrotIndustries, 2019-05-20:
Are you using https://github.com/horizon-eda/horizon/commit/dfa9fad3d884b31c1a987e3d99dd3cf84acdd9a6 or newer?

krasimirk, 2019-05-20:
How to find out?
There is no "About" dialog.
I checked out before one month from the git and compiled manually.

carrotIndustries, 2019-05-20:
It's at the bottom of the start screen (the one showing recent pools and projects)

krasimirk, 2019-05-20:
[image]
Does not seems like check sum for git revision.

carrotIndustries, 2019-05-20:
> Are you using dfa9fad or newer?

You're using a version older than that. That commit should have fixed this issue.

atoav, 2019-05-28:
If this has been already fixed, it can be closed. @krasimirk did it solve your problems?

krasimirk, 2019-05-28:
I will try tonight to upgrade to the latest sources and to check if it fixed.

atoav, 2019-05-28:
Ahh no pressure, just thought it might have been forgotten, because the last post was 8 days ago : )

krasimirk, 2019-05-28:
[image]
[image]
Not fixed yet.
Let me now if you want more information, debug info or whatsoever.
Regards,
Krasimir Kostadinov

carrotIndustries, 2019-07-01:
Sorry for letting this sit for so long, but that's likely caused by moving the pool repo to the new organisation. Either manually update the remote to point to the new repo or just re-download the pool and and copy your modifications to the new one.

carrotIndustries, 2020-10-08:
most likely fixed by now

closed 2020-10-08 by carrotIndustries