horizon-eda/horizon #271: macOS Support

phlmn, 2019-05-24:
I think it's a good idea to track the state of macOS support inside an issue.

I started to implement macOS support with #264, but are is quite some things missing. Things I currently have in my mind:

- [x] make it compile (#264)
- [x] make it start (#264)
- [ ] make the board viewer work
- [ ] fix icons

phlmn, 2019-05-24:
## OpenGL Issue (from #264)

Unfortunately the board viewer does not work yet. glLineWidth(…) and glBlitFrameBuffer(…) are causing an error. By removing glLineWidth(…) and drawing directly to the screen instead of using glBlitFrameBuffer(…) the board viewer does start. Debugging OpenGL is not much fun, maybe someone has an idea.

Some potentially useful apitrace output:

```
$ ./apitrace replay horizon-imp.5.trace
/Users/phlmn/Coding/apitrace/build/../../horizon/horizon-imp
1 @0 CGLRetainPixelFormat(pix = 0x7ff682f98870) = 0x7ff682f98870
1: warning: unsupported CGLRetainPixelFormat call
2 @0 CGLReleasePixelFormat(pix = 0x7ff682f98870)
2: warning: unsupported CGLReleasePixelFormat call
4 @0 CGLRetainContext(ctx = 0x7ff68406a400) = 0x7ff68406a400
4: warning: unsupported CGLRetainContext call
5 @0 CGLReleaseContext(ctx = 0x7ff68406a400)
5: warning: unsupported CGLReleaseContext call
6 @0 CGLReleasePixelFormat(pix = 0x7ff682f98870)
6: warning: unsupported CGLReleasePixelFormat call
14 @0 CGLRetainPixelFormat(pix = 0x7ff682e7a510) = 0x7ff682e7a510
14: warning: unsupported CGLRetainPixelFormat call
15 @0 CGLReleasePixelFormat(pix = 0x7ff682e7a510)
15: warning: unsupported CGLReleasePixelFormat call
17 @0 CGLRetainContext(ctx = 0x7ff68492f800) = 0x7ff68492f800
17: warning: unsupported CGLRetainContext call
18 @0 CGLReleaseContext(ctx = 0x7ff68492f800)
18: warning: unsupported CGLReleaseContext call
19 @0 CGLReleasePixelFormat(pix = 0x7ff682e7a510)
19: warning: unsupported CGLReleasePixelFormat call
GL_RENDERER: Intel(R) Iris(TM) Graphics 550
GL_VENDOR: Intel Inc.
477 @0 glLinkProgram(program = 8)
477: warning: WARNING: Output of vertex shader 'oid_to_geom' not read by geometry shader
WARNING: Output of vertex shader 'lod_to_geom' not read by geometry shader

509 @0 glLinkProgram(program = 9)
509: warning: WARNING: Output of vertex shader 'oid_to_geom' not read by geometry shader

541 @0 glLinkProgram(program = 10)
541: warning: WARNING: Output of vertex shader 'oid_to_geom' not read by geometry shader
WARNING: Output of geometry shader 'flags_to_fragment' not read by fragment shader

573 @0 glLinkProgram(program = 11)
573: warning: WARNING: Output of vertex shader 'oid_to_geom' not read by geometry shader
WARNING: Output of geometry shader 'flags_to_fragment' not read by fragment shader

GL_RENDERER: Intel(R) Iris(TM) Graphics 550
GL_VENDOR: Intel Inc.
1299 @0 glLineWidth(width = 2)
1299: warning: glGetError(glLineWidth) = GL_INVALID_VALUE
1311 @0 glLineWidth(width = 2)
1311: warning: glGetError(glLineWidth) = GL_INVALID_VALUE
Rendered 0 frames in 0.241093 secs, average of 0 fps
```

wraybowling, 2020-01-28:
Maybe move this to Milestones if it's a priority. https://github.com/horizon-eda/horizon/milestones

carrotIndustries, 2020-01-28:
I'm still a bit undecided whether it's worth given the current state of OpenGL and Gtk on mac os. There still seem to be quite a lot of glitches in GtkGLArea's quartz backend and apple's move to only allow notarized apps to run doesn't make mac os an attractive platform to target.

schneider-engineering, 2020-10-12:
@phlmn Icons are fixed by providing `hicolor-icon-theme`, e.g. through brew. The issue with `glBlitFrameBuffer()` is caused as the renderbuffer on the GLArea framebuffer ist GL_RGB8 instead of GL_RGBA8. `set_has_alpha(true)` somewhere in the constructor fixes this. I'm using GTK+3 stable (3.24.23), on the Master branch half of GLArea is rewritten, not entirely sure what problems will arise then. 

This is how horizon on 10.15 currently looks for me:
[image]
[image]

Main problem so far is window resizing. This is completely broken. Moving between HiDPI (internal) and normal DPI (external monitor is a problem as well, not entirely sure if this is related.

@carrotIndustries I get your point about notarizing, at least bundling an app package seems somewhat supported by the project itself: [GTK OSX Bundling wiki page](https://wiki.gnome.org/action/show/Projects/GTK/OSX/Bundling). In general I'm surprised how much actually works.

schneider-engineering, 2020-10-12:
FWIW, the resizing issues seems to be an GTK issue. Happens with the glarea demo as well. Looks pretty much like this is the same issue: https://gitlab.gnome.org/GNOME/gtk/-/issues/1298

carrotIndustries, 2020-10-12:
That's excellent news! To me that's enough to claim at least 'experimental' mac OS support. Please open a PR with the changes you did to make it build an run or tell me which of the existing mac OS-related PRs to merge. With that done we can also add CI to make sure it keeps building on mac OS.

schneider-engineering, 2020-10-12:
This was built using master, I will have a look at #290, which I missed so far, thanks for pointing that out.

Things I would like to hear your opinion on before creating a PR:
* `cppzmq` is kind of hard to get on OSX. Would you consider to bundle it? I just threw it into `3rd_party` for a first try.
* podofo doesn't play well with clang and c++17.
`/usr/local/include/podofo/base/PdfFilter.h:466:17: error: no template named 'auto_ptr' in namespace 'std'`. I fixed this by adding [`-D_LIBCPP_ENABLE_CXX17_REMOVED_AUTO_PTR`](https://libcxx.llvm.org/docs/UsingLibcxx.html#c-17-specific-configuration-macros) to `CXXFLAGS`. Not a real c++ dev here, so no clue what side effects could be caused by this. Opinions?

Disclaimers:
* Grid rendering is only sloppy fixed so far, by forcing the line thickness to 1. Fixing this by actually using quads *sounds* easy enough but isn't done.
* As of now there is NO support for anything related to the menubar (The menu macOS has at the top of the screen)
* Keybindings are untouched, Command+Q works on some windows, but even minimizing a window with default system keybindings won't work right now. Not a GTK dev here.
* There is no app bundling done yet. #290 has something in that regards, but as this is developer grade anyhow right now, I can live without an app bundle just fine. Window resizing has to be fixed before thinking about distributing this in an easier fashion anyhow.
* As mentioned: resize the window and you can basically restart horizon.

carrotIndustries, 2020-10-12:
> As mentioned: resize the window and you can basically restart horizon.

Oh, didn't realize it's that bad. Claiming mac OS support with that level of buggyness will most likely cause more annoyances than not supporting it for the time being.

> cppzmq is kind of hard to get on OSX. Would you consider to bundle it? 

Isn't it available in homebrew? For msys2 for example, I added the c++ bindings to the zeromq package upstream: https://github.com/msys2/MINGW-packages/pull/2193

> podofo doesn't play well with clang and c++17

We've had the same issue on freebsd, so adding darwin to that if should to the trick: https://github.com/horizon-eda/horizon/blob/c4b9f8f04819a55120e3c357139ce46c8c11a60c/Makefile#L734-L736

> Fixing this by actually using quads sounds easy enough but isn't done.

Another avenue could be to draw quads and turn them into grid marks in the fragment shader. My gut feeling tells me that his might be slightly faster.

> As of now there is NO support for anything related to the menubar 

I guess that this could alienate a lot of mac OS users, so we probably need to come up with one.

> but even minimizing a window with default system keybindings won't work right now. Not a GTK dev here.

If that also happens with other Gtk apps, you should probably report in the gtk bugtracker.

isometric, 2020-12-20:
Looks like GTK got a new OGL backend for macOS.

https://blogs.gnome.org/chergert/2020/12/15/gtk-4-got-a-new-macos-backend-now-with-opengl/

carrotIndustries, 2020-12-20:
> Looks like GTK got a new OGL backend for macOS.-

That's good news, but that's Gtk 4. It'll likely take quite some time till we use Gtk4 since it deprecated some features we need: https://discourse.gnome.org/t/using-gtkpopovermenu-as-a-gtkmenu-replacement/3786

We also need to make sure that a usable version of Gtk4 is available in all distros we want to support, i.e. Debian stable and Ubuntu LTS.

isometric, 2020-12-20:
RIP

isometric, 2020-12-21:
@schneider-engineering  Can you share your changes? For zmq, relying on osrf/simulation/cppzmq seems fine. 
@carrotIndustries  How set against maintaining 3+4 are you? Never done GTK stuff but might take a shot at it over the holidays, a few pointers would be helpful.

carrotIndustries, 2020-12-21:
> How set against maintaining 3+4 are you? 

As far as I can tell, that's not really feasible since there have been some significant API changes, such as `gtk_box_pack_start` being replaced by `gtk_box_append`. Also check out https://developer.gnome.org/gtk4/4.0/gtk-migrating-3-to-4.html Gtkmm4 also has just been released a day ago.

All in all, the earliest we can start thinking about Gtk4 is if it's available on all platforms as it's either gtk3 or gtk4.

schneider-engineering, 2020-12-21:
@isometric somehow lost sight of this. will have a look tomorrow, ok?

isometric, 2020-12-23:
It seems like longer term we should consider dropping from OGL to GLES and then maybe ANGLE if it comes to that. Seems like OGL all but dead as future support is concerned. If not for macOS it might then for the rPi 4 and successors.

carrotIndustries, 2020-12-23:
> It seems like longer term we should consider dropping from OGL to GLES 

That idea has crossed my mind as well. The minimum GLES version required though would be 3.2 due to extensive (or excessive?) use of geometry shaders. We should keep support for OpenGL proper as that allows for more optimal rendering such as `glDrawElementsInstancedBaseInstance` to draw all instances of one model in the 3D preview in one call.

My main motivation for supporting GLES is bringing the app to non-x86 platforms since none(?) of the GPUs in ARM SoCs support non-ES OpenGL.

isometric, 2020-12-23:
Hmm it seems like the rPi is already GLES 3.2 - it also supports OGL 2 and I think some of the nV Tegra stuff supports OGL 4.x but it's not exactly the most relevant platform these days.

isometric, 2020-12-29:
So I was able to get horizon running by pull just the GL fixes in #290 without the memory violations. Also managed to get GTK built from source so might take a stab at resizing next week. Not sure why but I noticed that the black frame/halo around schematic/pcb doesn't appear in the gtk3-demo, any ideas @schneider-engineering?

It also looks like GTK4 just uses ANGLE for compat so we may as well just use that directly if this doesn't work. @carrotIndustries any idea how much work is needed for GLES?

carrotIndustries, 2020-12-29:
> any idea how much work is needed for GLES?

I don't have any experience at all with GLES, but apart from some functions as the mentioned `glDrawElementsInstancedBaseInstance` most of Horizon EDA's OGL usage is probably available in GLES 3.2. Anything below that would require some dramatic changes due to lack of geometry shaders. As far as I got the ANGLE docs, GLES 3.2 is stil WIP.

atoav, 2021-08-10:
I'd like to teach horizon as part of our university course, and the lack of macOS support is as of now the only thing blocking me from doing so, as many of our students have Macbooks. Sadly I don't have a mac myself so I can't really look into it, but I am definitly following this development : )

bob-u, 2021-08-17:
Perhaps the best way forward would be to run Easy EDA on MacOS MoltenVK via Zink:

https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/7383

Zink is OpenGL implementation running on top of Vulkan or in case of Mac MoltenVK.

david12481632, 2021-09-17:
> Perhaps the best way forward would be to run Easy EDA on MacOS MoltenVK via Zink:
> 
> https://gitlab.freedesktop.org/mesa/mesa/-/merge_requests/7383
> 
> Zink is OpenGL implementation running on top of Vulkan or in case of Mac MoltenVK.

That looks promising

https://www.collabora.com/news-and-blog/blog/2021/06/14/zink-summer-2021-update/

tkornack, 2022-01-05:
I would love to hear if anything has changed here. I came to say that I was able to compile without any trouble on Apple silicon with just a set of homebrew libraries. The GL view crashes, as described above. 

There are a load of macOS Eagle users who need to bail out now that it's really dead and this really seems like the right place to land. Anyone interested in taking sponsored work on this topic?

sparkylein, 2022-05-23:
Are there instructions for compiling Horizon on macOS (Apple Silicon)?

Used the makefile as is but it doesn't work.

Here the error message:
`Package libzmq was not found in the pkg-config search path.
Perhaps you should add the directory containing `libzmq.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libzmq' found
Package libarchive was not found in the pkg-config search path.
Perhaps you should add the directory containing `libarchive.pc'
to the PKG_CONFIG_PATH environment variable
No package 'libarchive' found
 build/obj/src/util/uuid.o
src/util/uuid.cpp:3:10: fatal error: 'glibmm/checksum.h' file not found
#include <glibmm/checksum.h>
         ^~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [build/obj/src/util/uuid.o] Error 1`

Home-brew could not find libzmq and lib archive was installed with 'brew install libarchive' but it's not pulled in.

Also, glibmm/checksum.h is not found even so I can manually compile gtkmm code with g++ just fine.

carrotIndustries, 2022-05-23:
For zeromq, `zeromq` and `zmqpp` seem to be the right packagesin homebrew.

sparkylein, 2022-05-24:
Yes, I found out that for libzmq I had to install zmq

However, lib archive is still not pulled in.

BTW: sorry, somehow I didn't see that you already answered, so I edited my former post...

sparkylein, 2022-05-24:
I've added:

export PKG_CONFIG_PATH="/opt/homebrew/opt/libarchive/lib/pkgconfig"
export LDFLAGS="-L/opt/homebrew/opt/libarchive/lib"
export CPPFLAGS="-I/opt/homebrew/opt/libarchive/include"

into the terminal and now Horizon started compiling... came to a halt at:

-------------------------

 build/obj/3rd_party/router/router/pns_router.o
In file included from 3rd_party/router/router/pns_router.cpp:41:
In file included from 3rd_party/router/router/pns_node.h:31:
3rd_party/router/include/core/optional.h:4:10: fatal error: 'boost/optional.hpp' file not found
#include <boost/optional.hpp>
         ^~~~~~~~~~~~~~~~~~~~
1 error generated.
make: *** [build/obj/3rd_party/router/router/pns_router.o] Error 1

-------------------------------------------------

carrotIndustries, 2022-05-24:
> but came to a halt at

you're missing `boost`

sparkylein, 2022-05-26:
Homebrew was used and all dependencies are installed.
(created a shell script to pull in all proper dependencies with homebrew, I can post the shell script if needed/wanted...)

Got some files compiled, got stuck (file not found) at: 
'TDocStd_Document.hxx'
'boost/optional.hpp'
'BRepBuilderAPI_MakeWire.hxx'
'glm/glm.hpp'

All paths are set, even reset homebrews' symlinks with:
"brew unlink glm && brew link --force glm"

Looked through the horizon-eda issues and found out that a few years ago there were discussions about the build process:
[https://github.com/horizon-eda/horizon/issues/70](url)
[https://github.com/horizon-eda/horizon/pull/111](url)
[https://github.com/horizon-eda/horizon/issues/523](url)
Sadly, the other gentleman removed their horizon fork, so I cannot try/reuse their build file.

After this I got the Windows 10 machine out and tried to compile the Windows version.
Instructions from: 
[https://horizon-eda.readthedocs.io/en/latest/build-win32.html](url)
have been used and it compiled many files until the same thing happened: files are not found:
"undefined references to `horizon::trim(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >&)"
Which puzzles me because the trim function is an internal horizon function.

Anyway, if I have time this weekend I might engage my intel-Mac and my Linux machine and try to compile horizon on them.

I will not propose a new build system, because I do not care what is used.
What I care about is that it compiles, but this not only on the developers' machine.

Any help is appreciated for macOS and Windows.
And for Linux I would like to know which distro works/compiles best; don't care which one, Arch, Debian, Manjaro, Solus, etc. as I'm all familiar with and will install the best distro for horizon :D
Thanks.

sparkylein, 2022-05-27:
Hi Schneider-engineering,
yes, found out about cppzmq before I edited my last post, which solved a lot of issues.
Still fighting lstdc++fs and opencascade (can't find the lib; there are only dylibs in the lib dir).

Anyway, tried to use your patch file but I get this error message:
> patching file Makefile
can't find file to patch at input line 52
Perhaps you should have used the -p or --strip option?
The text leading up to this was:
> 
> |diff --git a/src/canvas/canvas_gl.cpp b/src/canvas/canvas_gl.cpp
|index 7dbb8905..dc09dee3 100644
|--- a/src/canvas/canvas_gl.cpp
|+++ b/src/canvas/canvas_gl.cpp
> 
> File to patch: ...

BTW: I used a clean Makefile from the repo.
Thank you for your input.

sparkylein, 2022-05-27:
Started to compile on macOS-x86_64 and ran into the same problem as with macOS-arm64:
-lstdc++fs is holding up that the compilation goes through.

What am I missing?

schneider-engineering, 2022-05-27:
@sparkylein sorry, for the late response. Not sure why the patch would not apply for you. Glad you got it compiled in the end. Clang/LLVM has `<filesystem>` support starting from 9.0, so the flag is not supported. When you read the patch, you can see I removed it as well.

Would you mind sharing your environment variables, that made the makefile patch minimal?

What machine do you have? Do you have a retina display? From what I remember, half of the issues in GTK are caused by DPI and screen/pixel coordinate conversion issues.

What patches to canvas_gl where necessary for you to get it to render?

sparkylein, 2022-05-28:
Compile time:

macOS Monterey 12.4 on 2016 MacBook Pro with intel-i5: 9:50 minutes
(sysctl -n hw.ncpu counts 8 cores: 4 cpu cores and 4 threads)

macOS Monterey 12.4 on 2021 MacBook Pro with M1-Max: 2:10 minutes
(sysctl -n hw.ncpu counts 10 cores: 10 cpu cores)

DIdn't think the difference would be that big...

sparkylein, 2022-05-28:
As schneider-engineering mentioned "set_has_alpha(true);" needs to be set otherwise the schematic and board windows will crash and not open.
I've placed it in canvas_gl.cpp line 45 just above "set_can_focus(true);"

What's really weird is that both windows (schematic and board windows) have a thick dark frame surrounding them.
If the window is resized, the black frame does not resize, but if the window is resized bigger than the black frame that window portion is hidden... weird.

Also in WIndows all shortcuts are Ctrl-Z, Ctrl-S, etc. on Mac it needs to move to Cmd-S, Cmd-S respectively.

sparkylein, 2022-05-29:
The horizon imp window is acting weird on this platform.
When the window is resized, the whole window is resized, not only the GL canvas, but also the UI itself making everything (buttons, sliders, everything) distorted.
Spent many hours trying to understand how OpenGL is used in horizon, with no avail.
Looked at kicad's canvas code (as kicad doesn't have this resizing problem) but didn't find much similarities between the two to figure out why horizon's canvas distorts everything while resizing the window. )
Therefore, the aspect ratio follows the window size, which also makes drawn circles ovals, and with this horizon unusable.
The buttons, sliders, etc. change their shape and location which makes the cursor unusable to click on any ui widget.
Having no comments in the source code is not helping...

schneider-engineering, 2022-05-29:
hey @sparkylein,

gtkmm (3) on mac os does not seem to be too mature.

Very minimal Demo based on https://github.com/GNOME/gtkmm/blob/gtkmm-3-24/demos/gtk-demo/example_glarea.cc:
[image]
[image]

It has the exact same issues. This is not due to the lack of comments in Horizon ;) Interestingly enough the drop shadows around the window renders just fine, but all other issues are exactly the same (Even without the GLArea). 

As I'm neither comfortable with GTK, nor with C++, this is where I gave up the last time (and just kept using KiCAD)

sparkylein, 2022-05-30:
@schneider-engineering, you are correct, if a simple gtkmm app has the same issues, then it it likely impossible to fix this downstream…

Nevertheless, documenting code is a good habit to have, especially in a complicated software package like horizon. Looking through the code and see the different kind of technologies that were used to accomplish this and the fact that this is the work of one person amazes me. Just wonderful!
I can see so much potential in horizon, that IMHO it would be a bummer if in a few years from now @carrotIndustries is still the only one… and because of this is burned out. It becomes easier if the burden of fixing/enhancing the code base, and writing documentation, rests on more shoulders. 

I spent over a decade on Linux, which is great, but wanted better hardware, than HP, Lenovo, etc. can offer, which made me move to a Mac. Therefore, I would love to see horizon work well on macOS :)

tkornack, 2022-05-30:
> As I'm neither comfortable with GTK, nor with C++, this is where I gave up the last time (and just kept using KiCAD)

I also gave up but I have had success running Horizon on virtualized linux in UTM on both amd64/intel and aarch64/M1.

It seems that a lot of these issues were recently addressed in the new macos gtk backend 4 [1], so I think we should wait until horizon is ported to version 4. That might have to wait until 2023 which is roughly when gtk4 [2] will appear in a stable Debian with the release of Debian Bookworm [3].

[1] https://blogs.gnome.org/chergert/2020/12/15/gtk-4-got-a-new-macos-backend-now-with-opengl/
[2] https://tracker.debian.org/pkg/gtk4
[3] https://en.wikipedia.org/wiki/Debian_version_history#Debian_12_(Bookworm)

sparkylein, 2022-05-30:
@tkornack, thank you for the info.
Found even more: [https://blogs.gnome.org/chergert/2022/02/](url)

Looking forward when horizon will use gtk4 :)

sparkylein, 2022-05-30:
BTW: homebrew has already gtkmm-4.6.
Just saying :P

carrotIndustries, 2022-05-30:
> Looking forward when horizon will use gtk4 :)

Don't hold your breath. Right now, gtk 4 makes it much harder to attain the user experience I want for Horizon EDA:

 - https://gitlab.gnome.org/GNOME/gtk/-/issues/3502
 - https://discourse.gnome.org/t/popup-menus-in-gtk4/6278
 - https://discourse.gnome.org/t/pointer-warping-on-wayland/9197
 - https://discourse.gnome.org/t/using-gtkpopovermenu-as-a-gtkmenu-replacement/3786
 - On i3, without composting menus look like this:

[image]

Porting Horizon EDA to Gtk4 would be a multi-month effort and it'll most likely have worse user experience than the current Gtk3 implementation.

sparkylein, 2022-05-30:
@carrotIndustries, thank you for the links.
It shows that you are way ahead of us, as you posted gtk4 issues already a year ago.
Oh well, this is quite sobering... so, no horizon-eda any time soon on macOS :(

sparkylein, 2022-05-31:
Is this a different gtk beast?
[https://gitlab.gnome.org/GNOME/gtk-osx/](url)
[https://gitlab.gnome.org/GNOME/gtk-mac-integration/](url)
[https://gitlab.gnome.org/GNOME/gtk-mac-bundler/](url)
It works with gtk2 and gtk3.

Obviously this gnome/gtk project started life as:
[https://sourceforge.net/projects/gtk-cocoa/](url)
(which works with gtk1)

Not sure if it is worth looking into this...

carrotIndustries, 2022-05-31:
> Is this a different gtk beast?

From a cursory glance, they're all about improved integration or packaging. None of them will fix the issues the mac OS backend has.

phkahler, 2022-08-12:
Re: GTK4,  FYI a handy way to track preparedness is here if you want to copy it:
https://github.com/solvespace/solvespace/issues/853

carrotIndustries, 2022-08-12:
> Re: GTK4, FYI a handy way to track preparedness is here if you want to copy it:

Solvespace is in a much better position for adapting to a new UI toolkit since it already supports three and doesn't contain that much UI and UI-specific code to begin with. See https://github.com/horizon-eda/horizon/issues/271#issuecomment-1141390647

gitfoxi, 2023-05-05:
While you're not holding your breath you can try it under Ubuntu, using Parallels on Apple Silicon. It mostly works and performance is great.

Build it from source.

3D viewer doesn't work :(

```
No provider of glDrawElementsInstancedBaseInstance found.  Requires one of:
    Desktop OpenGL 4.2
    GL_ARB_base_instance
    GL_EXT_base_instance
```

If you really want 3D viewer with terrible performance, run it like:

    LIB_GL_ALWAYS_SOFTWARE=1 horizon-eda

tkornack, 2023-05-05:
[UTM](https://mac.getutm.app) also works well; 3D works and is tolerable.

gitfoxi, 2023-05-06:
@tkornack I took your advice and gave UTM a try.

At first it had the same OpenGL problems I'd seen elsewhere.

I learned that UTM has recently implemented a new display driver virtio-gpu-gl-pci which uses the GPU and is very snappy but breaks Horizon. I was surprised that even the LIB_GL_ALWAYS_SOFTWARE workaround does not help like it does in parallels. You can configure UTM to use the old display driver, virtio-ramfb, and then Horizon works but the performance is only tolerable, not great.

carrotIndustries, 2023-05-06:
> No provider of glDrawElementsInstancedBaseInstance found.  Requires one of:

Interesting, that's the first GL implementation I've come across that doesn't support this function. Implementing a workaround that uses one draw call per component should be possible. It's nothing that I plan on doing, but if someone wants to do it, I'm glad to help.

gitfoxi, 2023-05-07:
VMWare Fusion claims to have OpenGL 4.3 so I gave it a shot. My 3rd trip through the Ubuntu installer in as many days.

Good news: 3D works perfectly!

Bad news: Nothing else works!

The schematic and layout tools just show the blue background with the light blue plusses. You can zoom in and out on the plusses, but nothing else displays.

I learned that getting OpenGL to work consistently across different GPUs and OSs is something people are very interested in, because of games. But it must be very hard to accomplish, though, and there's no easy solutions in the marketplace.

The internet recommends to use middleware instead, specifically Orge3d, https://www.ogre3d.org/about/features which does seem to have a story about integrating with GTK. https://wiki.ogre3d.org/GtkmmOgre

gitfoxi, 2023-05-08:
@carrotIndustries, I gave it a try.

Parallels supports OpenGL 4.0. The function that requires OpenGL 4.2 is glDrawElementsInstancedBaseInstance. OpenGL 4.0 can support the similar-sounding function glDrawElementsInstanced. So,

glDrawElementsInstancedBaseInstance -- 👎 
glDrawElementsInstanced -- 👍 

The difference is the final parameter called baseinstance. What does baseinstance do? I read the documentation 5 times and I have no idea. It's a mystery to me.

[image]

 https://registry.khronos.org/OpenGL-Refpages/gl4/html/glDrawElementsInstancedBaseInstance.xhtml

What's passed to baseinstance is something called, `idxs.offset`. I can't figure out what it's for. But I decided to see what happens without it.

```
-                    glDrawElementsInstancedBaseInstance(GL_TRIANGLES, it.count, GL_UNSIGNED_INT,
-                                                        (void *)(it.face_index_offset * sizeof(int)), idxs.n_packages,
-                                                        idxs.offset);
+                    glDrawElementsInstanced(GL_TRIANGLES, it.count, GL_UNSIGNED_INT,
+                                                        (void *)(it.face_index_offset * sizeof(int)), idxs.n_packages);
```

What happens is, some components appear in the wrong place.

[image]

If I could understand what `idx.offset` is doing as a `baseinstance`, maybe I could make progress.

But there's more problems. With this fix the 3D viewer does load and run under Parallels without error. However, it's a black screen  except for the axis which animate perfectly.

[image]

 I don't know how I can make progress against a black screen.

tkornack, 2024-04-28:
> Don't hold your breath. Right now, gtk 4 makes it much harder to attain the user experience I want for Horizon EDA

I'm curious if anything has changed about GTK4 support, given the time that's passed or your experience writing Dune 3D.