horizon-eda/horizon #276: Crash when dragging one square onto another onto another

atoav, 2019-05-27:
## Problem

Moving certain geometry on top of each other crashes the parent window of an editor e.g. with status -1073741819. I had this multiple times already in the schematic editor, only now managed to check how to boil it down to a minimum reproducible case.

## Steps to replicate

1. draw two squares besides each other with some space inbetween
2. take one square and move it onto the other square in such a way, that two of the sides are on top of each other

A vague unbased suspect might be the merging of junctions..

carrotIndustries, 2019-05-28:
Unfortunately, I can't reproduce this error. Can you share a test project and perhaps a short screen capture to help me reproduce this?

atoav, 2019-05-28:
Ok, I could reproduce it with:
- the latest build
- deleted startup folder
- community pool
- new empty project

This is as vanilla as it gets. Here you can see me drawing the two lines then moving one onto the other using <kbd>m</kbd>, at which point it crashes:  

[image]

I run on Microsoft Windows Version 1809 (OS Build 17763.503) with `0.9.0 Aurora Commit f5f307b 2019-05-26 18:44:52 +0200 warnings for duplicate refdes`

carrotIndustries, 2019-05-28:
Duh, I extrapolated that lines == net lines

closed 2019-05-28 by carrotIndustries

atoav, 2019-05-28:
Oh sorry for the miscommunication

carrotIndustries, 2019-05-28:
Anyhow, thanks for reporting this. Running into things like these within the first moments of using the schematic editor can really ruin one's overall impression.

commit 4070b584 (2019-05-28): update line and arc junctions on merge

fixes #276