horizon-eda/horizon #277: Shoving traces sometimes effects defunct extra vias

fruchti, 2019-05-28:
Sometimes™, shoving traces with vias duplicates some vias, so the vias left at the original position naturally are disconnected from the trace's net. The extra vias then have to be manually deleted (but after that, everything seems fine). I've run into this numerous times but couldn't find out when exactly it happens.

However, I was now able to make a minimal example for reproducing the issue: 
[Example project](https://github.com/horizon-eda/horizon/files/3229839/via-test.tar.gz)

[image]

Here, the problem only seems to occur when drawing both traces within one route tool invocation, however I've had this issue with vias of ”older” traces as well.

The recording above was done with the newest version (657388e) on Arch.

fruchti, 2019-05-28:
Looking at my GIF again I see that the trace I draw first is closer to pad 1 of the left transistor than the rules should permit (I left everything at the default 0.1 mm). Playing a bit around, this does not seem to happen every time I draw this trace. This is probably an unrelated issue, though.

atoav, 2019-05-29:
I had the too close trace problem also happening, managed to catch it before manufacturing, but wasn't sure when it occured, so I didn't suspect a bug – but in hindsight it could have been.

commit c13d8149 (2019-05-29): router: set via parent item

should fix #277
closed 2019-05-29 by carrotIndustries