horizon-eda/horizon #278: Feature request: Select all Parts of Same kind

atoav, 2019-05-29:
I'd find it tremendously useful if we could add actions to the Schematic and Board editor, to allow a quick selection of all parts:
- of the same kind
- with the same package

This would be extremely useful for all tasks where you have to replace all parts with a certain MPN or package without manually selecting them, which is slow and error prone.

I imagine it like this:
1. Select a part
2. Envoke the "Select all same Parts" or "Select all of same Package"-Action
3. Have a selection of all same Parts/Packages

What do you think?

If you have multiple parts selected on invokation, the resulting selection should contain all parts contained within that initial selection. If no parts are selected, the action shouldn't be available.

This could also be added to the Right Click-Context Menu for good measure.

carrotIndustries, 2019-05-29:
There's on issue we need to take into account: We currently can only select items that are on the active sheet since the current selection is stored in the canvas.

In the board editor, we don't have that limitation.

We could also make this a part of #171 .

atoav, 2019-05-29:
> We could also make this a part of #171 .

That would be extremely cool.

> There's on issue we need to take into account: We currently can only select items that are on the active sheet since the current selection is stored in the canvas.

I was thinking about this already, but I think per sheet would be fine for such an action. For the outliner, it might be better to make it a global thing tho..