horizon-eda/horizon #280: Version information in .hprj?

fruchti, 2019-06-13:
Since .hprj files are human-readable and rather short, it could be useful to have some information about the Horizon version used to create the project in them. For example, if an old, archived project could no longer be opened with the current Horizon version for some reason, it would be very useful to be able to compile a working Horizon. If the information is machine-readable, Horizon could check if a project was created with a newer version and warn the user information may be lost when editing/saving the project with the older, currently installed version.

For a human-readable field the format in the About dialogue should be sensible. An automated version check probably works best with the commit timestamp and hash?

carrotIndustries, 2019-06-17:
> An automated version check probably works best with the commit timestamp and hash?

That'd generate too many false positives I suppose. Storing the major/minor/micro version and do the check based on that one seems to be a better way.

fruchti, 2019-06-18:
> > An automated version check probably works best with the commit timestamp and hash?
> 
> That'd generate too many false positives I suppose. Storing the major/minor/micro version and do the check based on that one seems to be a better way.

You're right. TBH, I wasn't aware there was already a version number, since the repository lists no releases. The stored version information should contain the commit hash, though, for the slight chance that some corner case breaks without a version number bump.

closed 2020-10-18 by carrotIndustries

carrotIndustries, 2020-10-18:
We now have version information all pool/project files.

When opening a new file with an old version:
[image]

Before saving a file in a new version:
[image]

commit 9f5cc52a (2020-10-18): add version information to all project/pool files

closes #280