horizon-eda/horizon #281: [UX] Minor Improvment: Set Zoom to Fit when Schematic is opened

atoav, 2019-06-14:
This is just a minor thing, but I noticed that horizon has some sort of default zoom level it recalls once you reopen a project. Uppon opening a schematic I always hit <kbd>Home</kbd> right away – maybe it would be reasonable to do this automatically for all sheets?

This allows you to get a "overview" first and then you can zoom in onto the parts that interest you – so it fits with the cognitive flow.

carrotIndustries, 2019-06-16:
Have a look at that branch and see if it works for you.

The whole thing is a bit hacky since we can't really tell if that's the first size-allocate is the final one visible to the user.

atoav, 2019-06-17:
Hey cool, it sort of improved things, but it is not _quite_ there for some reason. Pictures speak more than a thousand words so see here:

| master behaviour | branch behaviour | after pressing <kbd>Home</kbd> |
| --- | --- | --- |
|  [image] | [image] | [image] |

The first one is how it looks when I open the project on the master build. The second one is how it looks when you open it on the new branch. The third one is how it looks when you zoom to fit.

We can leave it like that, it is already an improvement.

commit 682de802 (2019-06-17): imp: zoom to fit on open

should fix #281
closed 2019-06-17 by carrotIndustries