horizon-eda/horizon #283: Parts containing footprints of other parts

atoav, 2019-06-14:
I am currently creating parts for the legendary Cherry MX keyboard switch series and found myself in a odd situation: the switches have basically slots where you can add a LED, which means the LED pins have to be part of the Cherry MX package.

So I did the straightforward thing and created a Entity for a push button with a LED, so when you place the thing in your circuit, you can place the button and the LED seperately.

## Missing from BOM : (

So far so good. The problem is, that the LED is not necessarily part of the switch itself, so you would have to order it seperately, which means the LEDs will be missing in the BOM unless you remember them or note them down externally. 

## Solution?

I am not sure if this is a common thing (probably not), but maybe the idea I came up with could be useful for something else. It goes like this:
Users can add one ore more "sockets" to a part (a bit like in Diablo II) and they can then put another part into said socket. Parts that are placed into sockets are not shown in on the schematic and don't need to be placed on the board but show up on the BOM export. If one or more parts have a empty sockets a warning shows up on BOM export.

This might also be useful for parts that need additional stuff that also should go to the BOM like button caps, knobs for potentiometers, screws and nuts, stuff like that.

Maybe this is all to much for such a small edge case, but I wanted to note the idea down.

fruchti, 2019-07-29:
I don't think this specific case is common enough to justify a new feature. But a generic comment field for parts which would show up as a BOM column would be useful for a few cases, both during ordering and for assembly:

* Does not include the LED, has to be ordered and placed additionally (your case)
* Mounting nut and washer not included (reminder to order these for some connectors)
* Different part orientation than in the datasheet (e.g. for the WS2812B, which has a weird pin 1 by default)
* Component origin on PCB edge instead of connector centre of mass (for pick and place of SMD connectors)

All these are obvious to the creator of the part, but are important to clarify e.g. for an external PCBA service provider.

carrotIndustries, 2019-07-29:
> I don't think this specific case is common enough to justify a new feature. But a generic comment field for parts which would show up as a BOM column would be useful for a few cases

That was my thought as well. Might as well make these editable in the BOM export window so that users can add their own notes on a per-project basis.

fruchti, 2019-07-30:
> That was my thought as well. Might as well make these editable in the BOM export window so that users can add their own notes on a per-project basis.

Good idea. Manual changes to the comments after export would otherwise be lost or would have to be reapplied manually with each BOM update.

atoav, 2019-10-09:
I renamed this issue to reflect the new direction of the discussion