horizon-eda/horizon #286: Assembly view export

atoav, 2019-06-28:
Currently I am taking screenshots of the Assembly view and combine them to get a high resolution image which I then recolor and write a legend for.

I don't have to do that very often so it is not a pressing issue, but I know you have a export assembly view feature planed so I was curious about how it would look like?

Maybe for starters a `Export visible Layers as PDF` similar to the thing in the schematic editor would be totally enough?

carrotIndustries, 2019-06-30:
I've had something similar to altium's draftsman in mind for implementing assembly drawing export: https://www.altium.com/documentation/19.0/display/ADES/((Draftsman))_AD

However, since that's likely to be a significant effort, something a bit more basic seems to be a reasonable idea.

fruchti, 2019-07-02:
I've thought about this lately as well. I think the “Export visible Layers as PDF” option would be a very reasonable solution that would remain useful as a “quick export” even when a more powerful tool comes along.

carrotIndustries, 2019-07-10:
> However, since that's likely to be a significant effort, something a bit more basic seems to be a reasonable idea.

done in 1e1f308a1faa51541c2290c1afb081d2163cd5a4

fruchti, 2019-07-13:
This is great, thanks a lot!

atoav, 2019-07-14:
Very cool, works like a charm and does the job!

carrotIndustries, 2019-07-14:
Closing since everyone seems to be content with  1e1f308.

closed 2019-07-14 by carrotIndustries