horizon-eda/horizon #289: [UX] Feature Request: Indicate whether item is already selected in selection context menu

frmdstryr, 2019-07-15:
When I drag and select a bunch of components then attempt to deselect some, it's sometimes difficult to know which is selected or not when the selection menu pops up.  It would be nice if the menu would somehow indicate which item(s) are currently selected.

carrotIndustries, 2019-07-15:
Okay, I got something working that just appends `(selected)` to the menu item. Got any better idea?

frmdstryr, 2019-07-15:
That should work or maybe a [CheckMenuItem](https://developer.gnome.org/pygtk/stable/class-gtkcheckmenuitem.html) might look better if it's not hard to change to that?

[image]

closed 2019-07-15 by carrotIndustries

frmdstryr, 2019-07-15:
Works great, thanks!

commit 85c171a3 (2019-07-15): improve clarify selection menu

fixes #289