horizon-eda/horizon #291: Problem on Windows 7

Peta-T, 2019-08-18:
Just I run horizon-eda on W7 everything is OK any errors
but I do not see anything in scheme canvas, anything in board canvas and anything in 3D canvas.
Can somebody help me?
Thanks
[image]

carrotIndustries, 2019-08-31:
Sorry for letting this sit for so long, but that's very likely a driver issue. Gtk seems to be a bit picky when it comes to OpenGL drivers. IIRC Intel HD4000 graphics are known to not work on windows.

EdizonTN, 2019-09-02:
I have a same problem - ATI Mobility Radeon HD3650. :-(

carrotIndustries, 2019-09-05:
Even though not immediately solving the problem, try the latest build https://ci.appveyor.com/project/carrotIndustries/horizon/builds/27217664/artifacts

Open a schematic/board window, press <kbd>Ctrl+Shift+o</kbd> in the project manager window and paste the stdout/stderr output in here. That at least gets us closer to finding out what the actual issue is.

EdizonTN, 2019-09-06:
STDOUT:
`[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] imp rx false`

STDERR:
`[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:16.195: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:18.050: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:18.079: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:18.981: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.035: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.083: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.132: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.174: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.214: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.229: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.258: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.273: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.289: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.307: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.322: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.340: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.356: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.372: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.389: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:19.405: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:37:46.475: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.294: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.346: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.371: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.400: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.425: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.451: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.470: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:04.488: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:17.605: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:17.635: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:17.662: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:18.523: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:18.556: fb setup not supported
[... cut ...]
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:20.445: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:20.462: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:20.480: fb setup not supported
[C:\Users\Peter\Documents\Projects\Workspace-HorizonEDA\Projects\Test\top_sch.json] (horizon-eda:6408): Gtk-WARNING **: 19:38:20.497: fb setup not supported
`

carrotIndustries, 2019-09-06:
Seems like we're facing the issue described in https://gitlab.gnome.org/GNOME/gtk/issues/1003 and https://www.mikrocontroller.net/topic/417908#5431397 (TLDR: driver-related) The guy in the gtk issue and me did more or less the same thing to debug this issue without success. 

I'm afraid to say that there's not much I can do about this issue right now. The best advice I can give is using another computer with a newer GPU.

Peta-T, 2019-09-07:
I had also Gtk-WARNING output as EdizonNT
my Graphics adapter is Nvidia NVS 5400M on Lenovo T530 notebook
Autodesk Inventor work OK

Peta-T, 2019-09-07:
Its look like problem similar to this, but who is ble to repait it?

https://bugzilla.gnome.org/show_bug.cgi?id=792174
...

 paul84uk 2018-04-13 17:22:09 UTC

(In reply to Daniel Boles from comment #1)
> I get this on GTK+ 3.22.28 from MSYS2 on Intel HD Graphics 4000:
> https://bugzilla.gnome.org/show_bug.cgi?id=794283#c5
> 
> I have no idea whether it's correct; it seems reasonable to me to report if
> GTK+ can't do something it's asked to do... is there indication out there
> that this *should* work?

When I looked at bit deeper, opengl was reporting a GL_FRAMEBUFFER_INCOMPLETE_ATTACHMENT error but I couldn't find an obvious cause. I think it has something to do with the implementation of the Framebuffer object in the intel drivers, and specifically the Framebuffer EXT functions. The symptoms of the problem are definitely hardware dependent.

If I force the Framebuffer to use a texture for rendering by setting GtkGLArea::has_alpha to TRUE then I get something rendered to the GtkGLArea but there still seem to be some issues with the depth buffer / rendering order which I can't resolve. 

I then recompiled Gtk-3.22 but replaced the calls to the Framebuffer extension functions with their core equivalents (glGenFramebuffersEXT becomes glGenFramebuffers etc). The laptop is running OpenGL4.2 and so the framebuffer object is core. Now if I set GtkGLArea::has_alpha to TRUE, there are no problems at all, but if GtkGLArea::has_alpha is FALSE so a RenderBuffer is used, the GtkGLArea is transparent and nothing is rendered. If I use this modified code on a PC with other graphics hardware(Windows, ATI Radeon 5400, OpenGL4.4), both texture and renderbuffer methods still work without issue .

So I don't have a solution but I can at least get something to work by replacing calls to the EXT functions in gtkglarea.c with their core equivalents and forcing the Framebuffer to use a texture for rendering instead of a Renderbuffer by setting GtkGLArea::has_alpha to true. By making this change to gtkglarea.c though, I'm guessing I've lost compatibility with older OpenGL versions where the Framebuffer object wasn't core.

Another user of the software is having problems on Intel 4000 graphics hardware that are very similar to those described in the linked thread (Bug 794283, crashing when window minimised). We're going to see if this work around helps next week.

EdizonTN, 2020-07-10:
Any update?
Still not working...
I have tested on 5 PCs without success.

carrotIndustries, 2020-07-10:
> Any update?

Yes, windows 7 has fallen out of support by the vendor, so I don't support it either to make my life easier. If it works, fine, if it doesn't I don't care that much.

After all, this seems to be a GTK issue caused by buggy(?) proprietary gpu drivers. I usually try to fix bugs in my dependencies, but this one isn't really worth my time since I can't even reproduce it as I don't have hardware that exhibits this bug and it's probably related to proprietary GPU drivers.

If you want this to improve, get involved with the Gtk people in the issues mentioned above.

> I have tested on 5 PCs without success.

I've tested on 5 PCs as well and all of them hand no problems running Horizon EDA.

pwnorbitals, 2021-04-05:
Your point has been made clear, @carrotIndustries, and I think it makes sense. Seeing this thread is probably going nowhere, we can probably close it.

closed 2021-04-05 by carrotIndustries