horizon-eda/horizon #292: Feature Request: Add "Change Package" function for parts with alternate package

atoav, 2019-09-01:
I am currently changing my resistors to use alternate packages for the vertical parts. As I understand it the only reasonable way to assign the alternate package is by selecting the alternate package upon placement as seen below:  

[image]   

Two thoughts on this:
1. The feature is a bit to easy to overlook. Maybe we should also display alternate packages in some way on the left hand side in the table where you actually make the choice? Either as extra "flip out" rows or as some sort of indicator (E.g. a greyed out bold `(2)` in front of the Package.
2. It should be possible to change the package after placing it, possibly both from the schematic editor and the board editor.

carrotIndustries, 2019-09-01:
>  selecting the alternate package upon placement as seen below

Nope, this has zero effect on the package selected. Selecting a package in that menu just changes the preview below.

> Edit: I realized this is already possible from within the board editor. Maybe adding the same thing to the schematic editor would already be enough?

Yes, that's the only way to change the alternate package. Changing the package from the schematic is a bit problematic architecture-wise as the schematic then would need to know about packages.

atoav, 2019-09-02:
> Changing the package from the schematic is a bit problematic architecture-wise as the schematic then would need to know about packages.

Ah okay, I understand now. I think I misunderstood the whole thing. So this request makes no sense whatsoever : )

closed 2019-09-02 by atoav