horizon-eda/horizon #296: Language translation

Peta-T, 2019-09-25:
Hello, is it possible to make to translate to different language?

carrotIndustries, 2019-09-25:
No, translation isn't planned so far and kinda pointless in my opinion:
 - You'd need to translate documentation as well
 - Certain words are difficult to translate (keepout, diffpair)
 - A lot of work for very little gain

Peta-T, 2019-09-26:
It's a pity, because Horizon-eda is an amazing program and you don't have to translate it yourself. From your side is necessary only prepare program for  export texts  and other people can translate to their languages.
Thanks

ktrace, 2019-09-26:
There are several localization services, free for opensource projects:
https://weblate.org
https://www.transifex.com

On Thu, 26 Sep 2019 10:29:10 -0700
Peta-T <notifications@github.com> wrote:

> It's a pity, because Horizon-eda is an amazing program and you don't
> have to translate it yourself. From your side is necessary only
> prepare program for  export texts  and other people can translate to
> their languages. Thanks
> 

-- 
  С уважением,            |  Best regards,
          Виктор Кустов   |          Victor Kustov
  XMPP: ktrace@jabber.ru
  I use FREE operation system: 4.19.63-calculate GNU/Linux
  up 5 hours, 18 minutes

Peta-T, 2019-11-11:
There is Czech translation of manual:
https://github.com/Peta-T/Horizon-EDA_cs.pdf/blob/master/horizon-eda_cs.pdf

carrotIndustries, 2019-12-24:
As mentioned above, translating the app itself isn't worth it and won't be supported in the near future. Translating the docs is fine though.

closed 2019-12-24 by carrotIndustries