horizon-eda/horizon #297: Compiling on Debian 10

Peta-T, 2019-09-27:
Hello,
I have fresh installed Debian 10 and successful compiled Horizon-EDA
with this dependencies
su
apt install libgtkmm-3.0-dev
apt install libcairomm-1.0-dev
apt install librsvg2-dev
apt install util-linux
apt install libyaml-cpp-dev
apt install libsqlite3-dev
apt install libboost-all-dev
apt install libzmq3-dev
apt install libglm-dev
apt install libgit2-dev
apt install libcurl3-dev
apt install liboce-ocaf-dev
apt install libpodofo-dev
apt install libzip-dev
apt install git
exit

git clone http://github.com/carrotIndustries/horizon
cd horizon
make -j 4

it is OK

but  when I open my older project found mistake on top layer
[image]
with this error
[image]
how can I repair it
Thanks
P

carrotIndustries, 2019-09-28:
The issue isn't with debian 10. The library used for triangulation has the bad habit of crashing when provided with input it doesn't like such as duplicate points.

Have a look at the position mentioned in the log window and watch out for solder mask features that touch each other at exactly one point and the like.

Peta-T, 2019-10-06:
OK,
according coordinates in logs I found that if i remove 2 vias all is ok and if I put this vias again in litle different positions all is ok
Thanks.

[image]
[image]

[image]

closed 2019-10-06 by Peta-T