horizon-eda/horizon #300: update all planes before 3dpreview and fabrication output

Peta-T, 2019-10-06:
Hello,
it is my mistake, but when I did fabrication output of my project i forgot updated allplanes
and result is wrong PCB without planes on bottom side.
my suggestion is to prevent it will be better add procedure Update all planes
automatic before fabrication output and before 3d preview
Thanks
P

carrotIndustries, 2019-10-06:
Shouldn't this be solved by https://github.com/horizon-eda/horizon/commit/6c2543e41ef3333f26292e6e9230f17d4fdaba98?

Peta-T, 2019-10-07:
Yes this is right
[image]
but will be much more user friendly something like this (this is not right code)
imp_board.cpp 

```
   connect_action(ActionID::VIEW_3D, [this](const auto &a) {
+        core.b->get_board()->update_planes();
+        core.r->commit();
        view_3d_window->update();
        view_3d_window->present();
    });
```

sorry I am not able to change this functionality
Thanks

atoav, 2019-10-09:
So you are proposing a auto-update instead of a prompt? I wouldn't be against it.

Additionally we could automatically update the planes when the file is reopened – in the assumption that this is the most common usecase.

fruchti, 2019-10-09:
For me, personally, the warning dialogue introduced in 6c2543e41ef3333f26292e6e9230f17d4fdaba98 has always been enough, but I wouldn't be against making the warning more visible or automatically updating planes. It's also a good habit to check the exported files in a gerber viewer (I found lots of issues last minute by doing this).

If planes are automatically updated before opening the export window, there should some indication what Horizon is doing, however. I have layouts where updating all planes takes about seven seconds. For more complex boards or less powerful machines, it would seem like Horizon hangs for a while when opening the export window, for no visible reason.

Another approach would be to display the preflight check warning in the export window. There could be a checkbox you had to click to ignore the message and maybe even a plane update button. I'm not sure this fits into Horizon's UI concept, though.

atoav, 2019-10-09:
Thanks for the interesting input @fruchti I think from a pure usability/reliability standpoint you'd nearly _always_ like to update planes before exporting, so doing it automatically reduces cognitive overhead and makes the software easier to use. Are there any cases where you'd like to export a board that _has_ planes but without updating them first? I can't think about any case. If I don't like to export a plane, I would delete it.

So in other words: the reason why we do this manual plane update thing is purely one of performance. Of course it is also needed to edit the planes, but strictly speaking there is no real reason why this has to be linked to a manual update process. One positive about the manual process is that you see a "ghost" of where the plane previously had been, which can be useful at times.

My proposal would be to make the plane update on export automatic and just see it as yet another part of the export process, and show a progress bar for the whole thing with a message what is happening at the moment (e.g.: "Updating Planes"). If you want to export without planes, there could be a switch for this:

[image]

fruchti, 2019-10-09:
> Are there any cases where you'd like to export a board that _has_ planes but without updating them first?

I think I once did this deliberately in KiCad as a workaround for some missing functionality. The workflow was update planes -> modify some geometry -> export without updating again. I don't remember why exactly this was necessary but in any case this is a terrible hack and shouldn't be encouraged (especially since plane geometry isn't saved).

> My proposal would be to make the plane update on export automatic and just see it as yet another part of the export process, and show a progress bar for the whole thing with a message what is [...]

I like this very much! I'd label the switch ‘Don't update planes’, though. It would keep the planes as they currently are and not ignore them completely. If you didn't want the planes in the output for some reason, you could clear them manually in the board editor and workarounds like the one described above would still be possible.

atoav, 2019-10-09:
>  I'd label the switch ‘Don't update planes’, though. 

Right, that makes more sense.

carrotIndustries, 2019-10-11:
People already mentioned that updating planes can take a lot of time. Another issue with the plane update as it's implemented right now is that it's a tool and thus creates an entry in the undo/redo stack, thus doing it when preparing the 3D view would be a bit distracting.

The problem with updating planes on startup is that is that it'd make startup take much longer.

atoav, 2019-10-11:
IMO automatic plane updates before 3D-preview are not necessary – sometimes one might inspect the traces without having the planes in the way, so I think not updating before the 3D-preview is the way to go.

For fabrication output this is okay, because you normally don't export very often and I think the time lost is tiny in comparison to the anger one would feel when they accidentally get broken PCBs from the fab, because of a missing plane update.

For the startup thing, I get it. Is there a reason why the state of the planes is not stored on disk? Becasue this would feel like the natural way of things: when you reopen a file you will see it exactly the same way you left it the last time.

carrotIndustries, 2019-10-12:
731636fb4c8d0aaeafbc2787b91629f413ac92d7 saves planes to disk as well for my test project, the board file size increased from 2 to 6MB, but saving is still fast enough. If that'll ever become a bottleneck we can think about storing planes more efficiently.

atoav, 2019-10-12:
I just tried it and it seems to work quite well, at least I found no issues with it. That would resolve the startup thing.

Peta-T, 2019-10-23:
OK I have tried also and its look good.
Thanks you very much.

closed 2019-10-23 by Peta-T