horizon-eda/horizon #301: [UX/papercut] Make textfields not loose focus when sidepane is not hovered

atoav, 2019-10-09:
The textfields in the sidepane lose focus when the cursor is outside the sidepane, this leads to the annoying situation, that you click in the textfield, accidentally move the mouse aoutside, start to type and discover you are fireing one key shortcuts like wild.

This also kind of breaks with the conventions known from other software and the web, so a fix would improve beginner friendlieness.

*Expectation:* once a text field is focused via click, it should keep focus until:
- the mouse is clicked somewhere else
- the user presses <kbd>Enter</kbd> (which should return the focus back to the canvas)

fruchti, 2019-10-09:
This annoyed me too, until I got used to it. Even though I always set my window managers up so the window focus follows the mouse, this behaviour wasn't intuitive for me.

atoav, 2019-10-09:
> This annoyed me too, until I got used to it.  

Which classifies this perfectly as a little annoyance aka "paper cut". These things are not dramatic on their own and you can happily never fix them. But they add up over time and reduce the user experience – especially for new users. Once you got used to it, it is very easy to overlook them, which makes noting them down even more valuable.

carrotIndustries, 2019-10-09:
The canvas grabs keyboard focus as soon as the cursor enters it to make it easy to start entering shortcuts after having clicked on a button outside of the canvas since these grab keyboard focus as well when being clicked.

I'll look into if special-casing the property panel from this is feasible.

carrotIndustries, 2019-10-09:
See https://github.com/horizon-eda/horizon/tree/canvas_focus for my take on this: If the focus is inside the property panel, the canvas doesn't steal it by hovering.

Perhaps we should restrict this to text/numeric entries, as losing keyboard focus on a button in the property panel is acceptable.

atoav, 2019-10-09:
> Perhaps we should restrict this to text/numeric entries, as losing keyboard focus on a button in the property panel is acceptable.

Agreed – when a user clicks a text input they expect to be able to type away independent of the position of the mouse. I can't really think of other keyboard controlled parts that would catch it.

carrotIndustries, 2019-10-10:
See the branch for this. If this matches your expectations, I'll merge it.

atoav, 2019-10-11:
Matches my expectations : )

closed 2019-10-11 by carrotIndustries

fruchti, 2019-11-09:
I actually now run into the opposite issue: Thinking I had confirmed a value by pressing Enter while the input field actually still had focus and thus entering hotkeys there. Could the Enter and Esc keys be mapped for input fields? Enter would confirm the value and transfer focus to the canvas, Esc would restore it to the previous value and transfer focus to the canvas, too.

fruchti, 2019-11-12:
Esc now does what I'd expect, but Enter still doesn't focus the canvas for me.

Another thing: I often change A value with the scroll wheel while hovering over the input field. Right now, this focuses the input so I'll have to deselect it afterwards but in my intuition, it shouldn't. Maybe this is a personal thing, though.

carrotIndustries, 2019-11-12:
> A value with the scroll wheel while hovering over the input field. Right now, this focuses the input so I'll have to deselect it afterwards but in my intuition, it shouldn't.

That's a Gtk thing so there's not much we can do about it however. We could make the canvas steal the focus if spinbutton has focus. Right now the canvas doesn't steal focus as a spinbutton is a subclass of an entry. That however would make typing in spinbuttons a bit more awkward

atoav, 2019-11-12:
> We could make the canvas steal the focus if spinbutton has focus. Right now the canvas doesn't steal focus as a spinbutton is a subclass of an entry. That however would make typing in spinbuttons a bit more awkward.

Could one differenciate between _hover_ focus and _clicked_ focus? That would come close to the expected behaviour: if you hover and scroll the value should change as long as you hover and if you move away focus is lost. If you however click the field it catches the focus and keeps it until you escape or enter.

fruchti, 2019-11-12:
> That's a Gtk thing so there's not much we can do about it however.

I already expected something like that. It's probably best not to mess with the Gtk defaults too much, since they're there for a reason. I don't know if my intuition is the one of the majority here, after all.

commit 3f8162e9 (2019-10-11): don't steal focus from text input widgets in property panel

fixes #301