horizon-eda/horizon #302: [UX] Feature Request: Allow usage of sheet variables on board

atoav, 2019-10-09:
The idea is simple: we have sheet variables in the schematic editor – it would be cool if we could use them in text placed on the PCB.

*Example:* let's say I have a `$rev : "1.0"` in the schematic. If I now place a text containing `$rev` on the board or in the schematic it should expand to `1.0` automatically.

This could help to avoid accidental inconsistencies between schematic and silkscreen.

carrotIndustries, 2019-10-12:
Currently, title block values are stored in the schematic, so the board doesn't have access to them. Moving them to the netlist should be feasible.

If we have some kind of text variable substitution in place, what other uses could we come up with?

atoav, 2019-10-12:
> If we have some kind of text variable substitution in place, what other uses could we come up with?

Mostly things that you want to have on the board and on the schematic. So for example a URL to the repo of the open hardware project, an authors mail adress for support, the project title etc.

atoav, 2019-12-20:
So the current usecases for these variables (on both schematic and board) that come to my mind are:
- author
- project_title
- rev
- license

These are all things I can imagine on a board and it makes sense to always keep them synced to the schematic (e.g. if the rev changes in the schematic, this could change the $rev placed on the board as well).

My feeling is the best would be to just implement this in a generic way, so people can come up with uses themselves. That means: _any_ key used in the schematics can be used as `$key` on both the board and the schematic and expands to the entered value. I wonder if it could also make sense then to add simple `Add Text: project_title`, `Add Text: rev`, etc. actions for increased discoverability of the feature.

I will create another semi-related issue that discusses the whole meta-data-thing from a usability perspective. As it is now, it is not bad, but I think there are some edges that could be sanded down.

carrotIndustries, 2020-02-11:
With #330 done, we have the basis to deal with this issue. Should we try to substitute variables by default in all texts on the board or add an option to texts to enable/disable substitution?

atoav, 2020-02-11:
> With #330 done, we have the basis to deal with this issue. Should we try to substitute variables by default in all texts on the board or add an option to texts to enable/disable substitution?

I think substituting per default is fine, as long as $ without space after it is still a delimiter for it

closed 2020-02-16 by carrotIndustries

atoav, 2020-02-16:
Works like a charm! Thanks a lot!

commit c1293e25 (2020-02-16): replace project meta variables on board

fixes #302