horizon-eda/horizon #307: Feature request: Set holes NC or silence ‘withotu’

fruchti, 2019-10-31:
It can be useful to place a hole with a pad (or with plating), but without electrical connection to anything on the board. This is possible

carrotIndustries, 2019-11-02:
I added this warning since I didn't see any purpose in having copper on the board that isn't connected to any net.

What's your use case for floating PTH holes?

fruchti, 2019-11-02:
Fair enough. Floating PTH mounting holes could probably be used where some have connections and mechanical/aesthetical uniformity is important. Maybe this use case is too fringe to justify this feature, though.

In my specific case, I'm using holes to create a prototyping area, since that is the most flexible way to place an array of PTH pads on a board. Since opening the issue, I've found out I can create the pad stack as ‘mechanical’ to silence this warning, but a separate pad stack feels like more of a workaround to me than a way to set them NC.

BTW. I noticed creating a pad stack with copper pads and an NPTH hole effects a warning ‘NPTH hole() near PTH hole’ in the board editor. In my understanding, this should be possible to manufacture even without clearance between pad and hole. Again, the use case is pretty niche but I feel like a warning would be better suited at the pad stack level than the board level.

carrotIndustries, 2020-02-16:
How about making this something that'll get flagged in DRC instead of the obnoxious warning?

Rather than adding another rule this would be a checkbutton in the preflight check rule.

fruchti, 2020-03-07:
> How about making this something that'll get flagged in DRC instead of the obnoxious warning?

This sounds like a good idea. I'm note sure if it warrants its own checkbox, though. Would it be feasible to group DRC warnings in more detail than they are now? The results could look approximately like this:
```
+ Copper clearance
  + Track near pad
    …
  …
+ Preflight checks
  + Airwires
    …
  + PTH hole without connection
    …
  …
…
```
This would make it a lot easier to quickly sort through the messages and ignore irrelevant ones. IMHO, adding checkboxes for cases which you might want to ignore would be unnecessary with grouping like this.

> BTW. I noticed creating a pad stack with copper pads and an NPTH hole effects a warning ‘NPTH hole() near PTH hole’ in the board editor. In my understanding, this should be possible to manufacture [...]

I'm now getting ‘NPTH hole() near NPTH hole’ when a pad is near a NPTH hole (regardless of whether the hole and pad belong to the same padstack or not), which is pretty misleading.

carrotIndustries, 2020-03-17:
> The results could look approximately like this:

Adding another level of hierarchy would significantly complicate things as the result returned by checks currently looks like this: https://github.com/horizon-eda/horizon/blob/master/src/rules/rules.hpp#L19

The preflight checks are more of a catch-all for checks as adding a rule incurs quite a lot of overhead. Usually, rules should only check for one thing.

While we're at it, how about making everything in the preflight checks optional with a checkbox?

fruchti, 2020-03-18:
> > The results could look approximately like this:
> 
> Adding another level of hierarchy would significantly complicate things as the result returned by checks currently looks like this: https://github.com/horizon-eda/horizon/blob/master/src/rules/rules.hpp#L19

I see. Since you'd desire to solve all DRC issues anyway, the extra sorting would rarely be necessary. That energy is probably better spent on eliminating false positives (should they come up).

> While we're at it, how about making everything in the preflight checks optional with a checkbox?

I'm not sure it would really be useful. I haven't had a board not pass a preflight check which I kept that way. However, if you wanted to add checkboxes I'd like to see a warning ‘Some preflight checks have been disabled’ in the DRC check window (otherwise it's too easy to forget about disabled checks).

For the arguably special use case of a prototyping area I think it's fine to work around the DRC by using a mechanical pad stack. Since it's the only one I was able to come up here it's probably not worth introducing additional complexity for. I'm closing this issue for now but we can still reference to this discussion if another user trips over it.

closed 2020-03-18 by fruchti