horizon-eda/horizon #312: std::exception: database is locked when changing tabs in pool manager during pool update

fruchti, 2019-11-02:
Steps to reproduce:

1. Open a pool, click ‘update’
2. Change the tab while the pool update is running, e.g. to ‘packages’

A message box ‘Exception in signal handler’ appears.

carrotIndustries, 2019-11-02:
Unfortunately there's no real fix for this as we may not be able to use the pool database while the pool update is in progress.

Instead of the ugly exception window we now have something a bit more pretty:

[image]

closed 2019-11-02 by carrotIndustries

fruchti, 2019-11-02:
This is what I would expect in this situation anyway, so no issues. However, now Horizon completely locks up if you click ‘Pool Cache’ in a project while its pool is updating in a second window.

carrotIndustries, 2019-11-02:
> However, now Horizon completely locks up if you click ‘Pool Cache’ in a project while its pool is updating in a second window.

As far as I can tell that's always been the case. Solving that might be a bit harder than expected:

If sqlite finds that the DB is busy, it sleeps for 1s and then checks if the db is ready. If that's still not the case, an exception is thrown. However that sleep blocks the gtk main loop and unfortunately the  pool update thread sends its progress to the main thread and the main thread is unable to handle these messages (since it's blocked in sleep).

A possible fix would be to have sqlite return busy (throw the exception) immediately. This avoids the deadlock, but makes the "database busy" message show up more often while the pool is updating.

fruchti, 2019-11-02:
You're correct. I thought I had an error message box in that case before but I tried an earlier commit and it locked up as well.

> Solving that might be a bit harder than expected

A full pool update is only rarely necessary anyway so I'm not sure a fix would be worth it then.

carrotIndustries, 2019-11-02:
> A full pool update is only rarely necessary anyway so I'm not sure a fix would be worth it then.

Because of that the commit above implements this:

> A possible fix would be to have sqlite return busy (throw the exception) immediately. This avoids the deadlock, but makes the "database busy" message show up more often while the pool is updating.

The pool cache window now also fails a bit more gracefully than marking all cached items as missing. Not perfect, but good enough for a edge case that I believe few people will encounter.

fruchti, 2019-11-02:
Wow, thanks for the fast fix!

commit 1c14c5d9 (2019-11-02): pool browser: handle busy DB more gracefully

fixes #312