horizon-eda/horizon #313: Keepout display

fruchti, 2019-11-05:
Currently, keepout polygons are shown in exactly the same way as copper polygons which is pretty confusing, since they do the exact opposite. Ideally, their appearance could be set independently from the rest of the copper layer.

carrotIndustries, 2019-11-05:
TBH, this annoys me as well, but I haven't yet thought enough of a way to display them. Some suggestions:

 - Distinct border color
 - Checkerboard fill

Any ideas?

fruchti, 2019-11-05:
Would it be possible to merge the ‘Layers’ and ‘Options’ panels? Instead of just the layers there could be an expandable tree like this:

```
·  <o> [ ] Top Mask

+  <o> Top Silkscreen
|- <o> [_] Graphics
|- <_> [_] Text

+  <o> Top Copper
|- <o> [_] Keepouts
|- <_> [#] Planes
|- <o> [#] Tracks
```

You could toggle each item's visibility and choose the way it's displayed. You could even extend the ‘Select only on work layer’ checkbox to limit the selection to the selected element type.

This might be even more intuitive than the current way since the ‘Options’ panel is so rarely used that you might forget about it (it happened to me with KiCad that I wondered for a while why my silkscreen text was gone until I found I had deactivated it).

If you're looking for an easier solution, another idea would be the way KiCad shows unfilled planes (diagonal lines, but only near the edges):

```
+------------+
|////////////|
|/          /|
|/          /|
|////////////|
+------------+
```

I'm not sure if a checkerboard pattern wouldn't add to much visual noise. For me, keepouts aren't that important for an overview of a layout (globally) but rather when placing components/routing (locally). A lower-contrast indication would be sufficient IMHO as you wouldn't want all keepouts to be immediately visible when looking at the whole board.

carrotIndustries, 2019-11-05:
> You could toggle each item's visibility and choose the way it's displayed. You could even extend the ‘Select only on work layer’ checkbox to limit the selection to the selected element type.

That seems to be the right way long-term. I've had something similar in mind in https://github.com/horizon-eda/horizon/issues/159

>  Somehow unify layer display options and selection filter

The main reason that hasn't happened yet is that I haven't come up with a neat way to present this.

One way might be to add a "selectable" togglebutton right next to the visible one.

atoav, 2019-12-04:
> Would it be possible to merge the ‘Layers’ and ‘Options’ panels? Instead of just the layers there could be an expandable tree

I think from a UI-Standpoint it would make even more sense to keep it seperated. E.g. like this example, where the Layer `Top Mask` is currently selected:
```
Layers                     |
========================== |
[[<o> Top Mask ]]          |
<o> Top Silkscreen         |
<o> Top Copper             |
... more Layers            |
========================== |
                           |
                           |
                           |
Top Mask Options           |
========================== |
<o> [_] Graphics           |
<_> [_] Text               |
... more Details/Settings  |
========================== |
```
So depending on which layer is currently selected, you see it's options. That could become useful in the future as well (if more options need to be available per layer). Also it is visually more clear than a tree (which is usually always expanded when you don't want it expanded and always collapsed, when you want it expanded).

I wouldn't scrap a _global_ display filter for this tho. The best place for global display filters would be IMO a heads up element in the right top of the  IMP a bit like Blender does it:  
[image]

pwnorbitals, 2022-01-24:
I think this one can be closed no @carrotIndustries ! ;)

carrotIndustries, 2022-01-24:
Indeed.

closed 2022-01-24 by carrotIndustries