horizon-eda/horizon #316: make clean is not clean enough

steinm, 2019-11-13:
There are still some leftovers after 'make clean'. I suggest to add $OBJ_COMMON to the 'rm' in the clean target. But even then, there is still a __pycache__ dir and some *.d files in build/obj/.... I actually don't know how to get rid of.

carrotIndustries, 2019-11-13:
@endofexclusive would you see anything wrong with `rm -r`'ing the entire build directory instead of trying to delete each file individually?

endofexclusive, 2019-11-13:
I see nothing wrong with removing the entire build directory via `$(RM) -rf $(BUILDDIR)`.

The reason it was not done originally was basically to prevent bad things to happen in case there was a bug in the `Makefile`, or if someone assigns `BUILDDIR` in a creative way.

An alternative could be to have a dedicated target like `make cleaner` or `make distclean` that removes recursively.

steinm, 2019-11-14:
The background for this issue was creating the debian package which couldn't be built twice in a row. The second build failed because of those leftovers from the last run. I personally think it would be better to make the current clean target tidier.

PR #318 (2019-11-16): make 'make clean' really clean
fixes #316

@endofexclusive  does that look about right to you?
commit 28bf91af (2019-11-16): make 'make clean' really clean

fixes #316
closed 2019-11-16 by carrotIndustries