horizon-eda/horizon #319: Feature request: Cutouts in packages

fruchti, 2019-11-15:
Some parts need a PCB cutout. Currently, this has to be manually created in the board editor, but each package could just as well contain an ‘Outline’ layer (like it does in KiCad).

carrotIndustries, 2019-11-15:
I've thought of this as well, but there are some edge cases I haven't fully figured out yet.

Should there be something similar to smash slikscreen/omit silkscreen in case one needs to tweak the cutout?

What to to with overlapping cutouts? The easy way would be to union all cutouts and subtract these from the actual outline using clipper. However, we'd need to turn arcs to straight line segments which would land us at #257. Or is leaving them as-is acceptable?

fruchti, 2019-11-16:
An ‘omit cutouts’ option or ‘smash cutouts’ action seems very reasonable here. At first I thought this would be one of the cases where project-individual fine adjustments would be impossible by design (like pad sizes) but that could provoke the workaround of having a normal variant and a ‘(no cutout)’ alternate in the pool. Since overlapping cutouts would mostly occur when more care has to be taken for the mechanical design and you'd probably redraw them anyway, I think computing the union of all cutouts wouldn't be worth the effort and possible problems.

commit 03f29b57 (2020-03-21): add outline layer to packages

also make outline smashable

closes #319
closed 2020-03-21 by carrotIndustries