horizon-eda/horizon #322: Feature request: Inserting arcs in polygons

fruchti, 2019-11-16:
Related to #167: Currently, there doesn't seem to be a way to insert arcs into an existing polygon. I think the most intuitive way to have a ‘round off corner(s)’ tool that could be applied to polygon vertices. This would make it easy to round off corners (even selectively) of a rectangle or any more complex shape without redrawing it.

For a single vertex, it'd make sense to set the radius by mouse (similar to @atoav's animation in #167) so you can e.g. extend the arc right until a neighbouring vertex. If multiple vertices are selected, you'd have to select a reference point or input a corner radius directly.

carrotIndustries, 2019-11-25:
Instead of just implementing this tool, I decided to write a blog post on how to implement a tool: https://blog.horizon-eda.org/misc/2019/11/25/tool.html

The actual workings of the tool needed to close this issue should be pretty similar to the one used in the blog post, but the geometry calculations needed are expected to be a bit more involved including some edge cases.

atoav, 2019-12-04:
Sadly my C++ skills are practically limited to the C I know, so I can't help with that..

closed 2020-04-11 by carrotIndustries

fruchti, 2020-04-12:
Works perfectly, thanks!

commit b028dbe8 (2020-04-11): new tool: roundoff polygon vertex

closes #322