horizon-eda/horizon #323: Compiling fails on gnu hurd i386

steinm, 2019-11-25:
The debian build daemon for hurd-i386 reports an error because there is no implementation of std::string get_exe_dir() in src/utils/utils.c

 build/obj/src/util/placement.o
 build/obj/src/util/util.o
src/util/util.cpp:151:2: error: #error Dont know how to find path to executable on your OS.
  151 | #error Dont know how to find path to executable on your OS.
      |  ^~~~~

See also [logs of buildd for hurd-i386](https://buildd.debian.org/status/fetch.php?pkg=horizon-eda&arch=hurd-i386&ver=0.20191119-1&stamp=1574243764&raw=0)

carrotIndustries, 2019-11-25:
Apparently `dladdr` can be used to achieve something similar: https://www.gnu.org/software/hurd/hurd/translator/procfs/jkoenig/discussion.html#index7h1

However, is someone able to test if the resulting package works actually works as intended on hurd? I have doubts that there's a sufficiently well-working opengl implementation to use the interactive manipulator.

Would not supporting hurd be an acceptable option as well?

steinm, 2019-11-25:
If there is no way to make it work on hurd, then skipping that architecture would be option.

carrotIndustries, 2019-11-25:
> If there is no way to make it work on hurd, then skipping that architecture would be option.

I think I'd be able to arrive at a patch that compiles, but I don't see any time spent on this really worth it unless someone jumps in to actually test if the built package works as intended.

For now, I think not supporting hurd is the best way to go.

carrotIndustries, 2019-12-24:
> For now, I think not supporting hurd is the best way to go.

closed 2019-12-24 by carrotIndustries