horizon-eda/horizon #326: Compile error with Clang 8.0.1

endofexclusive, 2019-12-15:
Clang version 8.0.1 seems to have problems building some of the Horizon C++ files. FreeBSD 12.1 has Clang 8.0.1 as the default compiler.

FreeBSD 12.0 has Clang 6.0.1 as default compiler, Clang 6.0.1 built Horizon without issues.

```
$ uname -a
FreeBSD cl 12.1-RELEASE-p1 FreeBSD 12.1-RELEASE-p1 GENERIC  amd64

$ c++ -v
FreeBSD clang version 8.0.1 (tags/RELEASE_801/final 366581) (based on LLVM 8.0.1)

$ git describe --tags
v0.9.6

$ gmake QUIET=
mkdir -p build/obj/src/pool/
 build/obj/src/pool/pool_manager.o
c++ -c -Isrc -I3rd_party -g3 -D_USE_MATH_DEFINES -DGLM_ENABLE_EXPERIMENTAL -fdata-sections -ffunction-sections -I/usr/local/include -I/usr/local/include/uuid -I/usr/local/include/gtkmm-3.0 -I/usr/local/lib/gtkmm-3.0/include -I/usr/local/include/atkmm-1.6 -I/usr/local/include/atk-1.0 -I/usr/local/include/glib-2.0 -I/usr/local/lib/glib-2.0/include -I/usr/local/include/glibmm-2.4 -I/usr/local/lib/glibmm-2.4/include -I/usr/local/include/sigc++-2.0 -I/usr/local/lib/sigc++-2.0/include -I/usr/local/include/giomm-2.4 -I/usr/local/lib/giomm-2.4/include -I/usr/local/include/pangomm-1.4 -I/usr/local/lib/pangomm-1.4/include -I/usr/local/include/cairomm-1.0 -I/usr/local/lib/cairomm-1.0/include -I/usr/local/include/cairo -I/usr/local/include/pixman-1 -I/usr/local/include/freetype2 -I/usr/local/include/libdrm -I/usr/local/include/libpng16 -I/usr/local/include/pango-1.0 -I/usr/local/include/fribidi -I/usr/local/include/harfbuzz -I/usr/local/include/gtk-3.0 -I/usr/local/include/gdk-pixbuf-2.0 -I/usr/local/include/gio-unix-2.0/ -I/usr/local/include/at-spi2-atk/2.0 -I/usr/local/include/at-spi-2.0 -I/usr/local/include/dbus-1.0 -I/usr/local/lib/dbus-1.0/include -I/usr/local/include/gtk-3.0/unix-print -I/usr/local/include/gdkmm-3.0 -I/usr/local/lib/gdkmm-3.0/include -I/usr/local/include/librsvg-2.0 -D_THREAD_SAFE -pthread -I/usr/local/include/pgm-5.2  -MP -MMD -pthread -Wall -Wshadow -std=c++14 -O3 src/pool/pool_manager.cpp -o build/obj/src/pool/pool_manager.o
In file included from src/pool/pool_manager.cpp:5:
In file included from 3rd_party/nlohmann/json.hpp:53:
3rd_party/nlohmann/detail/conversions/from_json.hpp:288:9: error: static_assert failed due to requirement 'sizeof(nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long, unsigned long, double, std::allocator, adl_serializer>) == 0' "could not find from_json() method in T's namespace"
        static_assert(sizeof(BasicJsonType) == 0,
        ^             ~~~~~~~~~~~~~~~~~~~~~~~~~~
3rd_party/nlohmann/detail/conversions/from_json.hpp:303:16: note: in instantiation of function template specialization 'nlohmann::detail::from_json_fn::call<nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long, unsigned long, double, std::allocator, adl_serializer>,
      std::__1::basic_string_view<char, std::__1::char_traits<char> > >' requested here
        return call(j, val, priority_tag<1> {});
               ^
3rd_party/nlohmann/adl_serializer.hpp:26:9: note: in instantiation of function template specialization 'nlohmann::detail::from_json_fn::operator()<nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long, unsigned long, double, std::allocator, adl_serializer>,
      std::__1::basic_string_view<char, std::__1::char_traits<char> > >' requested here
        ::nlohmann::from_json(std::forward<BasicJsonType>(j), val);
        ^
3rd_party/nlohmann/json.hpp:2474:36: note: in instantiation of function template specialization 'nlohmann::adl_serializer<std::__1::basic_string_view<char, std::__1::char_traits<char> >, void>::from_json<const nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long, unsigned long, double,
      std::allocator, adl_serializer> &, std::__1::basic_string_view<char, std::__1::char_traits<char> > >' requested here
        JSONSerializer<ValueType>::from_json(*this, ret);
                                   ^
3rd_party/nlohmann/json.hpp:2735:16: note: in instantiation of function template specialization 'nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long, unsigned long, double, std::allocator, adl_serializer>::get<std::__1::basic_string_view<char, std::__1::char_traits<char> >,
      std::__1::basic_string_view<char, std::__1::char_traits<char> >, 0>' requested here
        return get<ValueType>();
               ^
/usr/include/c++/v1/string:888:29: note: in instantiation of function template specialization 'nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long, unsigned long, double, std::allocator, adl_serializer>::operator basic_string_view<std::__1::basic_string_view<char,
      std::__1::char_traits<char> >, 0>' requested here
        {__self_view __sv = __t; return assign(__sv);}
                            ^
src/pool/pool_manager.cpp:30:10: note: in instantiation of function template specialization 'std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >::operator=<nlohmann::basic_json<std::map, std::vector, std::__1::basic_string<char>, bool, long, unsigned long, double, std::allocator,
      adl_serializer>, void>' requested here
    name = j.at("name");
         ^
1 error generated.
gmake: *** [Makefile:745: build/obj/src/pool/pool_manager.o] Error 1
```

The full list of files failing to build is:
```
 build/obj/src/pool/pool_manager.o
 build/obj/src/block/block.o
 build/obj/src/schematic/schematic.o
 build/obj/src/schematic/sheet.o
 build/obj/src/pool/pool_parametric.o
 build/obj/src/preferences/preferences.o
 build/obj/src/util/stock_info_provider_partinfo.o
 build/obj/src/pool-prj-mgr/pool-prj-mgr-app_win.o
 build/obj/src/pool-prj-mgr/pool-mgr/pool_remote_box.o
```

GCC 9.2.0 can compile the file above by replacing `c++` on the command line with
`g++9`. (It does not link however, probably because all libs in the system are
built with Clang.)

```
$ g++9 --version
g++9 (FreeBSD Ports Collection) 9.2.0
```

carrotIndustries, 2019-12-15:
Fixed in https://github.com/horizon-eda/horizon/commit/63c0d774df72d5aa54bec64ec80b311abafed511 by upgrading the json library. To prevent this from happening again, we now have CI for freebsd as well: https://cirrus-ci.com/github/horizon-eda/horizon

closed 2019-12-15 by carrotIndustries

endofexclusive, 2019-12-15:
I tied the horizon-test-project with 63c0d77 right now and it works very well.
Thanks!