horizon-eda/horizon #327: Description in org.horizon_eda.HorizonEDA.metainfo.xml is a bit short

steinm, 2019-12-16:
Just a minor thing. According to https://appstream.debian.org/sid/main/issues/horizon-eda.html the description is somewhat short.

carrotIndustries, 2019-12-16:
Yeah, that's why the flathub PR still is WIP. Feel free to make suggestions for a more elaborate description.

steinm, 2019-12-16:
From the documentation of horizon-eda

Horizon EDA is a Electronics Design Application similar to tools like KiCAD or LibrePCB. It allows you to draw schematics, layout circuit boards and comfortably manage a parts collection.

carrotIndustries, 2019-12-16:
I'm not particularly happy with that description either as it references other applications. The description should be self-contained without comparing to other apps.

steinm, 2019-12-17:
How about:

Horizon EDA is an Electronics Design Application. It covers the complete design flow from schematic entry to gerber export and includes a unified editor for parts, the schematic and board.

carrotIndustries, 2020-02-11:
got fixed in ce364699521e23a5d772b056b459dac81ebc8d17

closed 2020-02-11 by carrotIndustries