horizon-eda/horizon #330: [UX] Ideas for Project Metadata

atoav, 2019-12-20:
# Project Metadata: the Status Quo

What I call "Project Metadata" is the information one can enter about the project. This happens currently in two places:
#### 1. when creating the project

Here we can set `project title`, "sluggified" project title used to name the folder:
[image]  

#### 2. In the Schematic Properties Window 

Here we can enter Key-Value pairs that get used by the sheets

[image]  

This is a proposal on how to (potentially) unify these two areas of the software. I very much like the idea of the Key value-pairs used in the schematic properties window, but I feel that given it's importance it is in a bit of an murky corner and maybe it would help to move it front and center.

# Proposal

## Don't hide it
  
The question came up when I thought about why the project title I enter when creating a new project isn't automatically used as a key-value pair in the Schematic Properties. My idea was to make the most common metadata fields part of both the "New Project" and the "Project Manager Windows". This would go hand in hand with the idea proposed in #302 (which would allow users to use text variables like `$rev` or `$project_title` both on the board and on the schematic), and makes much more sense in terms of placement – it is about the project after all.

So in my proposal you should be able to see important metadata fields on one glance from the project manager: Project Title, Author, Revision, Date, License – maybe something like this:

[image]  

The design is a bit utilitarian, but it gives the Project Manager Window more power in a good way. When a design change feels like you untangled a knot, it might be worth considering.

#### What changed

- Project Directory is now on top to give a clearer (=faster) visual seperation to the pool part.
- added a `Open Directory` Button to the Project Directory
- added a heap of metadata fields. Date has a `Update` Button to set it to the current date, Revision has buttons to change versions quick (although this would depend on the version format used, so it might not be a feasible solution).
- moved the pool path a bit up to unite it visually with the pool section more

#### What might be missing

- I didn't really think about how to implement the arbitrary key-value pairs currently possible in the schematic properties editor. Maybe a extra `Project Metadata` Button?

## The New Project-Dialoge

If these changes are compelling in the case of the Project Manager, this should be even more true in case of the New Project Dialoge – here we create our projects for the first time after all! The only real difference here is that the user needs to browse for a path and additionally enter a "shorthand" version of the title. 

I would move the Project Title to the top and automatically "sluggify" the title to a filename by switching it to lowercase, removing spaces, splitting it where it makes sense etc (e.g. "My Project - a cool project" becomes "my-project"). If users don't like what they see, they can still change it, but my guts feeling is that the project title should be first.

## On Defaults

I was also thinking about how to handle default values. Maybe it is too much to add a preferences page where you can define defaults – what about just storing the last thing you used when creating a project and using that? That would be flexible enough, saves time and gets out of the way.

carrotIndustries, 2019-12-24:
The reason why we currently can't implement this as suggested is that the project information fields that are editable in the schematic editor are stored in the schematic file itself (or in the netlist if we want to use these on boards as well). As of now, the schematic and board editor have no knowledge about the project file.

The project manager itself only deals with the project file. Having the project manager modify the netlist/schematic as well isn't really feasible without ugly workarounds such as telling the schematic editor to modify the netlist acoordingly.

Having the schematic/board editor know about the project file appears to be the better solution overall and I should have exactly this gathering dust on a branch somewhere...

atoav, 2019-12-24:
> Having the schematic/board editor know about the project file appears to be the better solution overall and I should have exactly this gathering dust on a branch somewhere...

That _does_ indeed sound like the cleaner solution. I guess the biggest challenge keep backwards compatibility for projects that already store this data within the schematic. But it sounds like you are more or less on board with the whole idea?

carrotIndustries, 2019-12-24:
>  guess the biggest challenge could be keeping backwards compatibility for projects that already store this data within the schematic.

The easiest way would be for the project manager to import the data from the schematic if it's there.

> But it sounds like you are more or less on board with the whole idea?

Yes, as this information is more specific to the project itself rather than the top level schematic.

I'm still not quite sure about what to do about sheet variables.

atoav, 2019-12-24:
> I'm still not quite sure about what to do about sheet variables.

Me neither. I kinda like the whole idea, but I don't really see why it should be limited to the schematic. I think in this case using a project based scope would be appropriate.

How about adding a sort of <kbd>+</kbd> and <kbd>-</kbd> button in the project manager, that allows you to add/remove arbitrary or custom key-value pairs. For all sheet variables that don't fit the planned Metadata fields they could be added as such additional fields.

The only thing we'd really loose that way is the scope (and maybe a little space).

atoav, 2019-12-25:
I made two mockups to clarify my idea a little bit. What do you think?

## Inital view without custom metadata fields

[image]  

## View with custom metadata added

[image]

carrotIndustries, 2019-12-25:
ok, that was roughly what I had in mind as well. I think we can leave the sheet variables as is in case someone wants to override these on a per-sheet basis.

Nevertheless, all that probably won't make it into the 1.0 release.

atoav, 2019-12-26:
> I think we can leave the sheet variables as is in case someone wants to override these on a per-sheet basis.

Makes sense.

> Nevertheless, all that probably won't make it into the 1.0 release.

A pity, but understandable considering the effort. I'd really like to come to FOSDEM, but I am afraid I have my exams right at that time.

carrotIndustries, 2020-02-02:
Now onto the implementation: since we want to have this information readily available in board and schematic, the best place to store this would be the netlist (block). However, the netlist isn't editable by the project manager since it's written by the schematic editor.

Right now, the schematic/board editor don't know about projects, but I have an old branch related to panlisation that made schematic/board editor load the project file. That way we could use the schematic editor to copy the project info into the netlist. Unfortunately, that'd mean that getting the project metadata from the project manager to the board would become a multi-step process:

1. edit project title
2. click save in project manager
3. click "reload metadata from project" in schematic editor
4. save schematic
5. reload the netlist in the board editor

We could also have both schematic and board editor both individually loading the project data into the schematic/boad, but that way the user could unintentionally end up with different versions of the project metadata in schematic vs board.

Any other suggestion how we could make this a smooth process?

atoav, 2020-02-02:
> Right now, the schematic/board editor don't know about projects

Okay I get it: schematic and board are currently seperated from the project and if I read your description right, there is no way for the 3 of them to "know" if the other changed.

I know stuff like this can get hairy pretty quickly, but lets ignore it for a minute and first ask what the desired result would be ideally. In the second step we can see if there are ways to bend the existing things into shape. I am speaking of course from a position of ignorance (both regarding the hairy details and C++), which is the only way I can really help you right now anyways.

## In an ideal world™

In an ideal world the metadata from the project would always be the source of truth. That means the moment it changes the others would see it. But given the existing decoupling of the elements, I suggest something else: there is no way for a Schematic or Board Editor window to be opened without an according Project Manager window anyways, so the places in which automatic checks for changes would be necessary are quite limited: whenever a Schematic or Board Editor window regains focus or when the <kbd>Save all</kbd> button has been clicked in the Project manager (whichever is easier to implement). In order to change a metadata value one has to have the window of the Project manager opened at some point, there is no way around it.

## Implementation
Note, that I don't really know all too much on how this is done now (which makes it *much* easier for to speak), but to me it is quite clear that the Board/schematic have to know about the project in some way. The Project manager should be the only place with mutable access to its metadata (while it is focused), while all the others have read access (while they are focused) and make sure to know about changed values (so they need to know when their own cached values are outdated). The rules of ownership¹ are not hurt here, as only one window can be active at one time anyways – so there should be a ways to implement this in a threadsafe way.

The question whether these changes should be read automatically, or whether they should be cached locally and the user has to click a <kbd>update Project Metadata [changed]</kbd> button is a question of style and philosophy. I could imagine use cases where one would want to change the project version but not the board version or so (e.g. to indicate that you didn't get around implementing changes yet), but maybe it is not needed. I could totally live with such a button provided it indicates that something has changed (I'd wish we had an indicator like this for the netlist in the Board view as well).

If this would be the goal, it would probably be the best solution to have Schematic and Board Editor handle this independently. So instead of having `Project ← Schematic ← Board` having it like `Project ← Schematic` and `Project ← Board` makes more sense like you said:  

> We could also have both schematic and board editor both individually loading the project data into the schematic/boad, but that way the user could unintentionally end up with different versions of the project metadata in schematic vs board.

This is only a problem if this is *invisible to the user*. If there is a way of implementing read access to the project metadata, having it check and visually indicate it shouldn't be too big of a problem unless I get something horribly wrong. However: if you have to read the value automatically anyways to check whether it changed, you could just update it automatically as well, so I guess the hardest part is actually checking for changes and how to interface these changes with the existing code.

---
¹ Rules of ownership: you are allowed to either have one or more (read) references OR one mutable (write) reference, but not both at the same time. This idea comes from Rust, but works well in other languages too.

carrotIndustries, 2020-02-03:
I've thought a bit more about this and having the project metadata editable in the project manager and using them in placeholders in schematic and board would incur a lot of complexity. Since my goal for this project is to keep implementation complexity low, I propose this:

 - Drop project name/title from project file and move all project metadata to the top block (netlist).
 - All places that currently get this from the project data will get this from the top block. (actually, only a few places in the project manager)
 - The project metadata will appear read-only in the proect manager and a button will take the user straight to the to-be-prettified schematic properties dialog (or a new one) that'll look somehow like the mockup in the initial comment. The project manager will automatically update the project metadata every time the schematic is saved.

atoav, 2020-02-04:
I understand, I was of course speaking with the freedom of someone who asked himself "how should it be" rather that "what could even be given the existing architecture of things". The project manager would be the natural place for this information to be in from a interface standpoint, however as someone who has written some software myself I know that just adding something on top can dramatically increase complexity and I agree that we don't want that (and refactoring the whole architecture would only make sense if there is more than one reason for it).

I think your proposed solution is close enough in terms of functionality and has the potential to be future proof.

Another consideration: I remarked in issue #282 that it is a bit annoying to set this information (rev, date, author, ...) every time one starts a new project, maybe this is the time to add some sort of defaults to the preferences?

carrotIndustries, 2020-02-09:
See https://github.com/horizon-eda/horizon/tree/prj-meta for what got implemented so far. Might need to add a message that tells the user to open and save the schematic one for the metadata to be moved to the netlist.

atoav, 2020-02-09:
Seems to work quite nice, maybe a small text on the bottom of the window (_"Please save the schematic to apply the metadata"_) would already be enough?

carrotIndustries, 2020-02-10:
> Seems to work quite nice, maybe a small text on the bottom of the window ("Please save the schematic to apply the metadata") would already be enough?

I explicitly wanted to have an 'in-your-face annoying' dialog to make sure that users actually do the update:

[image]

closed 2020-02-10 by carrotIndustries

fruchti, 2020-02-11:
Minor Papercut: When opening the schematic after this message appears, the save button is still disabled, so hitting Ctrl-s and closing the schematic again doesn't actually update the metadata. It would help if the schematic editor would consider the metadata update an unsaved change and immediately enable the save action.

carrotIndustries, 2020-02-11:
> It would help if the schematic editor would consider the metadata update an unsaved change and immediately enable the save action.

Thanks for the nudge. I initially noticed this as well, tried something to fix it and it didn't work so I hoped that no one will notice...

However, we now have a tiny bit of code that'll only run more or less once, but will stay in the codebase probably forever. That's why I didn't pursue this issue further on the first try.

carrotIndustries, 2020-02-11:
A nice side effect of resolving this issue is that we now finally have sensible window titles!

commit ef6f6472 (2020-02-10): unify project title and title blocks

project metadata (author, title) etc is now stored in the netlist rather
than in schematic and project file

closes #330