horizon-eda/horizon #332: [UX/Papercut] Colorschemes don't get an extension on export

atoav, 2019-12-21:
Not sure if this is intended behaviour, but when I export a color scheme the save dialoge won't add a file extension automatically. This is unfortunate because the import dialoge explicitly wants a `.json` and just won't show anything else.

I suggest appending `.json` to the chosen filename if it doesn't end in `.json` already.

Also: the internal color schemes use `.schematic` and `.board` – is there a rationale for not using this here?

carrotIndustries, 2019-12-23:
The extension for the builtin color schemes isn't that important after all. The color preset files get processed by this script: https://github.com/horizon-eda/horizon/blob/master/scripts/make_color_presets.py that strips everything after the first dot, so `.board.json` would be an option as well.

closed 2019-12-23 by carrotIndustries

atoav, 2019-12-23:
Understood! 

> so .board.json would be an option as well.

Sounds perfect to me

commit a1892fa9 (2019-12-23): preferences window export: automatically add extension

fixes #332