horizon-eda/horizon #333: [UX] Part Browser is hard to resize. Default Window size might be to big

atoav, 2019-12-21:
This is one of these issues I had for quite a while, but only now without a multi-monitor setup it got annoying enough for me to write an issue about it.

## Problem

The Part Browser is hard to resize, because (it seems) some columns seem to block the scaling, which makes it impossible for me (on a 1980px wide display) to see the preview pane on the right. There are people out there with even less wide displays, who might have even more problems with this. I think this is definitly something that should be fixed for the release if possible, because it is one of the first areas of interaction for any user that tries something out.

[image]

↑ Looks like a slapstick I know...

## Expected behaviour

If I scale the window it should just scale, no matter what. It shouldn't really matter if one can see all the columns. The columns should be able to be scaled freely without minimum width.

Additionally you might want to check the default window size for it, because it might be a tad too big for many systems.

carrotIndustries, 2019-12-23:
Do you happen to have some parametric data with particularly long values hidden on the other pages? The minimum size of the left pane is defined by the largest page in the notebook (tabs) to the left.

Do you have a copy of that pool available somewhere?

atoav, 2019-12-23:
It happend with the work branch of my pool: https://github.com/atoav/horizon-pool/tree/work

In any way, I don't think it is a good idea to let the content dictate the width of the window – the two solutions would be hiding the overflow or wrapping the text. Wrapping the text has the downside of creating unevenly high rows, which might be visually undesired. Hiding the overflow has the downside that it would make certain long things unreadable on small screens.

carrotIndustries, 2019-12-24:
Does https://github.com/horizon-eda/horizon/tree/part-browser-sizing fix things for you?

atoav, 2019-12-24:
That works already a lot better : ) Could solve the Problem IMO

carrotIndustries, 2019-12-24:
merged this branch. Is there anything left to do to close this issue?

atoav, 2019-12-24:
Can be closed I think

closed 2019-12-24 by atoav