horizon-eda/horizon #334: Handle read-only $HOME better

rroohhh, 2020-01-10:
For reasons my `$HOME` is readonly, so when I start horizon I am presented by this error message:
```
$ ./build/horizon-eda
terminate called after throwing an instance of 'Gio::Error'
Aborted
```

While one can figure out the reason for this error (for example by using `strace`), it would be nice if the error message would be a bit more descriptive and include what actually went wrong.

carrotIndustries, 2020-01-12:
Please run horizon-eda in gdb and tell it to break on exceptions (`catch throw`) to figure out where exactly it fails.

rroohhh, 2020-01-12:
Ah sorry should have included that information from the beginning:

```
(gdb) bt
#0  0x00007ffff6399240 in __cxa_throw () from /gnu/store/347y0zr1a9s2f5pkcncgi3gd0r33qq81-gcc-9.2.0-lib/lib/libstdc++.so.6
#1  0x00007ffff7710157 in Gio::Error::throw_func(_GError*) () from /home/robin/.guix-profile/lib/libgiomm-2.4.so.1
#2  0x00007ffff75a46f4 in Glib::Error::throw_exception(_GError*) () from /home/robin/.guix-profile/lib/libglibmm-2.4.so.1
#3  0x00007ffff7716c8e in Gio::File::make_directory_with_parents() () from /home/robin/.guix-profile/lib/libgiomm-2.4.so.1
#4  0x00000000005f43e9 in horizon::create_config_dir () at src/util/util.cpp:266
#5  0x00000000004abaa7 in main (argc=1, argv=0x7fffffffd968) at src/pool-prj-mgr/pool-prj-mgr-main.cpp:11
```

So it fails while trying to create the config directory.