horizon-eda/horizon #338: Supporting the Specctra interface

cortesis, 2020-02-01:
It would be nice, if Horizon EDA would support the Specctra interface (export as DSN files and import of Specctra Session files) such that externa autorouters like https://freerouting.org/ could be used to route the PCB.

carrotIndustries, 2020-02-01:
I'm afraid, but I'm not a big fan of auto routers, see https://horizon-eda.readthedocs.io/en/latest/non-goals.html#autorouter, so don't count on me to add support for the specctra interface.

If someone comes up with a PR that doesn't add too much complexity and has a pretty UI, I might consider merging it.

cortesis, 2020-02-04:
Ok thanks,
If I somehow find time in the next year(s), maybe I'll implement it.