horizon-eda/horizon #341: Compressing parts folder of pool

fbruetting, 2020-02-02:
v1.0.0

The official pool file count is extremely large (17.326 objects, with 15.734 residing in `parts` which only are 14,3 MB). This lengthens copy/delete/move/backup actions pretty much (they some people still don’t have SSDs today). Can you thus compress the `parts` folder in e.g. a TXZ/TGZ/TAR archive?

atoav, 2020-02-02:
Just as an aside, there also has been discussion regarding size in the [issue 131 of the horizon-pool](https://github.com/horizon-eda/horizon-pool/issues/131). The work branch of my own fork of the pool has 31710 objects, which is a glimpse of the future.. So looking at ways of deal with this more efficiently are certainly something that might become important sooner or later.

carrotIndustries, 2020-02-03:
Just compressing the files isn't feasible since we want them to be plain text/json for convenient version control. Having a file for each item is a feature.

fbruetting, 2020-02-03:
Maybe for developers who work on those files, but this creates a LOT of overhead for every user. Maybe these can be compressed by default, but with a “developer mode” switch made available as regular files?

Some file managers also are able to open archives like regular folders, not sure which programs could do this with which formats though…

atoav, 2020-02-03:
I think pretty much any file manager nowadays can browse the contents of a zip file, but I imagine there are other issues as well. Pool updates (and if you create parts and contribute them to the pool) are managed via git. And I am not sure if git repositories can work inside a compressed file..?

fbruetting, 2020-02-06:
Yeah, I meant that the parts you actively work on could be “checked out” – i.e. put as plaintext files in a folder like currently done. But all the other files (so usually 99%) could stay in such an archive.

Alternatively mounts could maybe be used. Those mounts are easily accessible, but on the disk they reside as just one big chunk.

katyo, 2020-10-28:
Because __git__ uses _zlib_ to compress data, as an option pool manager can extract data directly from bare __git__ repo using _libgit2_. In my opinion this is quite easy to implement.