horizon-eda/horizon #342: [UX] Proposal for improved selection filter usability

atoav, 2020-02-02:
The current selection filter implementation is extremely flexible, and really shines when the filter you want to configure has to be quite granular. For day-to-day applications (like select everything on the `Top Copper` layer it is a bit unwieldy tho – the steps to do this would be:  

1. Open Selection filter
2. Unfold Lines
3. Doubleclick Top Copper
4. Unfold Arcs
5. Click Top Copper
Unfold Texts
7. Click Top Copper
8. Unfold Polygons
9. Click Top Copper
10. Unfold Tracks
11. Click Top Copper 

Even if we assume the Selection filter window has been opened and the tree has been unfolded we are speaking of ~5 clicks through a nested hierarchy, that won't fit on my 4K display without scrolling. So the cognitive load is real here.

This list is great for filtering by element type (e.g. lines, arcs, etc), but this comes at the cost of the other use case where one wants to  filtering by layer.

## Proposal

[image]  

My proposal isn't meant as a replacement of the existing (granular) selection filter, but rather as an addition that supplements it. The idea is to have selectability toggles in the layer list, similar to the existing visibility toggles, that allow to check or uncheck single layers very quickly. Compared to the existing solution this has following advantages:
- limiting selection to one or more layers is very quick
- ability to see which layers currently have selectable objects right in the side bar

## Details
However the interaction with the existing selection filter needs to be defined. I suggest we do it this way:  
- both filters toggle essentially the same flags: the layer selection filter toggles the selectability of a single layer for *all* possible elements to the opposite of what is currently displayed. This is the equivalent of toggling all unfolded layer checkmarks in the granular selection filter
- if you have a more granular selection filter set (e.g. only `Tracks` on `Top Copper`) and you toggle this layer in the layer selection filter, the granular selection filter gets overriden and  all kind of elements on `Top Copper` are selectable
- if you have certain layers selectable in the layer selection filter and e.g. only want to select `Text` elements on these, you just open up the granular selection filter and toggle all other elements off
- if you toggle layers from the granular selection filter they get updated in the layer selection filter accordingly

## Display
This results in four classes of selectability in the layer filter:
- selectable (everything on a layer is selectable)
- limited (only certain elements on a layer are selectable)
- unselectable (nothing is selectable, but the layer is visible)
- hidden (nothing is selectable, layer is invisible)

Whether this means we need four different icons to represent these four different states, is a bit muddy to me. Maybe like this: 
- Selectable can be a mouse arrow
- Limited can be a mouse arrow with an astersik to indicate the presence of more granular filtering
- Unselectable is a crossed out arrow
- hidden can be either no icon at all or a grayed out arrow

carrotIndustries, 2020-02-11:
To me at least thet selection toggle in the layer list modifying buttons in the selection filter is a bit too much unexpected spooky action at a distance since there's no obvious (visual) link between those two.

At the very essence of things, the selection filter is a matrix with the layers on one axis and the object types on the other. Anyhow, I don't think that presenting this matrix to the user is something we should probably do as it'll contain many blanks (i.e. no packages on slikscreen layer) and will probably too big on screen.

atoav, 2020-02-12:
The link between the two must not be visual, it must be _semantic_. So if the user is operating in the context of "I want to filter things" they will understand – the thing in the main window is always parent of the thing in a seperate window. That is why I proposed the astersik as a symbol of advanced (more granular filtering), so you know about what is going on, even if you forgot about the other window you opened a while ago.

I am sorry, but I am convinced that selection filtering as it is now is really not up to task..

carrotIndustries, 2020-02-12:
> I am sorry, but I am convinced that selection filtering as it is now is really not up to task..

Don't get me wrong, I'm on your page as well in that regard.

How about instead adding a second column to the selection filter dialog with a checkbox for each layer?

atoav, 2020-02-12:
> How about instead adding a second column to the selection filter dialog with a checkbox for each layer?

Thought about this as well – maybe this is a good compromise – let's go for it

thagotaberry, 2020-02-18:
I'd prefer to have all selection filters in one place as well - which would be the selection filter tool. I was thinking that maybe there aren't actually too many different real scenarios. So rather than giving the user a pile of options which aren't needed in reality (matrix would be the extreme here) nor spreading the settings over different places (overall clarity would pay heavily for that convenience imho), the selection filter should be fine tuned to be handy in all the real life use cases, that occur frequently enough that 'low level' filter fiddling wouldn't cut it.
Is a whole column needed? Would someone want to select only the Top Paste? Or Bottom Courtyard? Or would an additional "Top/Bottom/All (etc.) Copper" checkbox suffice?

carrotIndustries, 2020-03-01:
@atoav any comments on listing all layers vs. a list of presets for common tasks?

atoav, 2020-03-01:
My idea was basically a mixture of the two. 

The Selection Filter Window is divided into two panes. In the left pane we have a list of all the layers  and in the right pane a list of all the type

A picture probably says more than a thousand words:
[image]  

If I click on `All Layers` in the left pane, I can narrow down the selectable types on all layers on the right side. If I then want to _additionally_ narrow it down (e.g. by excluding `Text` on `Top Silkscreen`) I can click `Top Silkscreen`  and change it in the new right pane. If you just want to filter out by layer, click the checkbox instead.

Meta-Accessors like "All Top Layers" or "All Bottom Layers" could be added as well.

This has some neat advantages:
- allows for all the filter can do now
- no unfolding (less screen space and clicks wasted)
- layer based filtering is easy and fast
- quite intuitive (?)
- covers most filtering needs one could ever encounter

I don't know GTK well enough to draft it up in a realistic fashion, but you might know how it could work out.

Pitrsek, 2020-04-04:
Interesting idea with the combined selection filter. Usually layer settings and selection filters are separated(in the stuff I've been working in -which does not make that approach the optimal/the only one that is correct). 
No matter how the selection/filtering is done, it would be great if we can save the layer view/selection filter(multiple configurations), and recall it with short cut