horizon-eda/horizon #349: Delete pools option in startup menu

greenscreenflicker, 2020-03-03:
There is missing a delete option on startup for the pool manager.
See image attached. I propose a rightclick on pool, to to delete those.

![grafik](https://user-images.githubusercontent.com/17407241/75801214-0c7b7700-5d7b-11ea-9719-d5b928f1afd7.png

carrotIndustries, 2020-03-03:
Are you referring to deleting the pool folder from disk or just removing them from the recent pools list?

greenscreenflicker, 2020-03-04:
I would propose to remove them from the pool list. Then a dialog pops up asking if you want to remove it from disk.

carrotIndustries, 2020-03-22:
rather then offering a option to delete things from disk, I added a menu item to open the pool/project in the file browser

closed 2020-03-22 by carrotIndustries

greenscreenflicker, 2020-03-24:
@carrotIndustries Lukas, thanks, I will test soon and give feedback.

greenscreenflicker, 2020-03-24:
@carrotIndustries Works! Thanks for the implementation.

commit 040bdc7a (2020-03-22): add context menu to recent items

fixes #349