horizon-eda/horizon #351: Autogenerator for Symbols from Units

anuejn, 2020-03-17:
Hey,
It would be great to have an option to autogenerate simple Symbols from units.
My use-case are especially FPGA io banks with many ios. There are often many of them and creating the Symbols by hand is repetitive and tedious. Having this automated would be nice :).

carrotIndustries, 2020-03-18:
97540cac0ce71fd23438de45bfe9da97e4a4ff54 adds an action to place all remaining pins in the map pin tool (press p)

anuejn, 2020-03-19:
thats already nice :)

carrotIndustries, 2020-03-19:
So what else is left to be automated?

rroohhh, 2020-03-20:
Well drawing a box that fits all the pins, adding the pins automatically, determining the side they are from their function (input, output or bidirectional) and adding `$REFDES` and `$VALUE` texts in "standard" places would be great I think.

carrotIndustries, 2020-03-20:
> determining the side they are from their function (input, output or bidirectional)

I don't think that this makes a lot of sense. I've seen symbols where people did exactly this and didn't think any further and thus ended up with `SDA` one one side and `SCL` on the other. Either the symbol is just a row of pins (FPGA IO bank) or some human thought is required to produce something useful.

rroohhh, 2020-03-20:
The way its done in other tools is usually some kind of dialog, that lets you choose the specifics of the pin placement. 

There could be a default button, that automatically assigns the pins to left, right, top, bottom according to their function. Additionally the user could be able to manually specify the location for each pin (maybe with some kind of list interface, such that selecting and assigning a side to many pins simultaneously is possible)

carrotIndustries, 2020-03-20:
24a8f4be5a340825a35d71a37609999c0cbe0828 adds an overlay that helps with aligning pins

> but generating a "ok" symbol and then tweaking it a bit is a decent workflow I think.

My feeling is that I'll probably spend more time cleaning up what the generator did than drawing it from scratch.

> Additionally the user could be able to manually specify the location for each pin 

How's that different from moving the pin? OTOH, I could certainly envision a tool that tries to be smart and moves overlapping pins out of the way.

To sum it up: Don't hold your breath for me implementing a generator.

rroohhh, 2020-03-24:
> 24a8f4b adds an overlay that helps with aligning pins

Neat, I have to try that!
> How's that different from moving the pin? OTOH, I could certainly envision a tool that tries to be smart and moves overlapping pins out of the way.

Not that different to moving the pin, the main difference would be automatic aligment / sizing of the symbol.
> To sum it up: Don't hold your breath for me implementing a generator.

Understandable, would you be opposed to having one in general, or would you be willing to merge one if somebody else were to implement one (altough I can't promise I will work one one soon).

carrotIndustries, 2020-03-29:
21763df7d0d46a4a55b7f1a154684fe489cd5371 adds a tool that helps with resizing box-style symbols