horizon-eda/horizon #354: Exception is signal handler error

Pitrsek, 2020-03-23:
Hi,
I've created a new unit in pool, saved it. Then I've realized that I made a mistake in file name. I made duplicate with correct file name, and I wanted to delete the original one in the pool. That's when(while deleting from the eda) error window appeared:
[image]

After the error window, the program crashed.   
Windows 10 64bit Build 18363
1.0.0 Aurora
Commit 3f777bd 2020-01-29 19:14:09 +0100 version 1.0.0

carrotIndustries, 2020-03-23:
Is this error reproducible? This could happen if there was a pool update running.

Pitrsek, 2020-03-23:
Indeed, when pool update is running, funny things happen. I was able to somehow reproduce it. I was doing the same with entities(copying, renaming, deleting), when I clicked to units tab and double click on unit, I got the same error.  
I also got a new one:
[image]
Double click on unit I have just deleted while pool update was in progress(the unit entry was still visible, as pool update has not updated it yet). When I did it again I got a proper message about file not being available

carrotIndustries, 2020-03-23:
Try the latest build https://ci.appveyor.com/project/carrotIndustries/horizon/builds/31666639 (when it's done, takes a while). To me at least, this completely fixed all brokenness during pool update.

Pitrsek, 2020-03-27:
I got the same error with a new version when I used it first time. As I'm working with the same parts, maybe pool was corrupted from the previous crash?
[image]
I haven't had any issues after pool rebuild/update, but I did not had much time to test it(I've just created one component and proceed to schematic).

carrotIndustries, 2020-03-27:
You'll need to run the pool update at least once to apply the database fix.