horizon-eda/horizon #357: Cannot export pick and place file: Error Output file not opened

atoav, 2020-03-24:
I can't export a pick and place file for some reason.

In the file chooser dialogue I can only select a directory, while the window title clearly tells me to select a file:  
[image]  

When I try to export it then tells me this:  
[image]  

I tried all possible options and also tried entering a path manually, but nothing seems to work.

carrotIndustries, 2020-03-24:
You didn't enter a filename for the top side, that's probably why. The title for the filechooser is set incorrectly, it indeed picks a directory.

atoav, 2020-03-24:
> You didn't enter a filename for the top side

Spot on – That Error message mislead me

carrotIndustries, 2020-03-24:
24281f3b3b19965507e7ec6c3285a79f8cb3854a fixes the filechooser's title

atoav, 2020-03-24:
> we should probably deactivate the export button until all entries are filled in

Excellent idea : )

carrotIndustries, 2020-03-25:
> we should probably deactivate the export button until all entries are filled in

done in eb2698bb55bff8605d53eeb6ea97b7e5e86e4f7f. The subsequent commits did the same for all other export windows.

closed 2020-03-25 by carrotIndustries

atoav, 2020-03-25:
Thank you very much