horizon-eda/horizon #358: Segfault while importing python module

iovaris, 2020-03-26:
**Git Hash**: 7a43a357f200d97418d1af6e0c5d93c96957fb04
**OS**: Ubuntu 18.04.4
**Python**: 3.6.9 (64-bit)

I'm hitting a segfault while trying to use the Python module detailed here https://horizon-eda.readthedocs.io/en/latest/python.html

1) cloned the source code, installed dependencies
2) ran `make -j 4`
3) ran `make build/horizon.so` (note: I had to add build/ to the path, a slight discrepancy with the docs. Could this be related to the issue?)
4) Created a test.py file with `import horizon` inside the build directory
5) ran `python3 -v test.py` 

segfault:
```
Python 3.6.9 (default, Nov  7 2019, 10:44:02)
[GCC 8.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
# extension module 'readline' loaded from '/usr/lib/python3.6/lib-dynload/readline.cpython-36m-x86_64-linux-gnu.so'
# extension module 'readline' executed from '/usr/lib/python3.6/lib-dynload/readline.cpython-36m-x86_64-linux-gnu.so'
import 'readline' # <_frozen_importlib_external.ExtensionFileLoader object at 0x7f7c26fa61d0>
import 'atexit' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.6/__pycache__/rlcompleter.cpython-36.pyc matches /usr/lib/python3.6/rlcompleter.py
# code object from '/usr/lib/python3.6/__pycache__/rlcompleter.cpython-36.pyc'
import 'rlcompleter' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c26fa62e8>
>>> import horizon
# /usr/lib/python3.6/__pycache__/re.cpython-36.pyc matches /usr/lib/python3.6/re.py
# code object from '/usr/lib/python3.6/__pycache__/re.cpython-36.pyc'
# /usr/lib/python3.6/__pycache__/enum.cpython-36.pyc matches /usr/lib/python3.6/enum.py
# code object from '/usr/lib/python3.6/__pycache__/enum.cpython-36.pyc'
import 'enum' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c26fae7f0>
# /usr/lib/python3.6/__pycache__/sre_compile.cpython-36.pyc matches /usr/lib/python3.6/sre_compile.py
# code object from '/usr/lib/python3.6/__pycache__/sre_compile.cpython-36.pyc'
import '_sre' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.6/__pycache__/sre_parse.cpython-36.pyc matches /usr/lib/python3.6/sre_parse.py
# code object from '/usr/lib/python3.6/__pycache__/sre_parse.cpython-36.pyc'
# /usr/lib/python3.6/__pycache__/sre_constants.cpython-36.pyc matches /usr/lib/python3.6/sre_constants.py
# code object from '/usr/lib/python3.6/__pycache__/sre_constants.cpython-36.pyc'
import 'sre_constants' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c268d9208>
import 'sre_parse' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c26fc8898>
import 'sre_compile' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c26fbea90>
# /usr/lib/python3.6/__pycache__/copyreg.cpython-36.pyc matches /usr/lib/python3.6/copyreg.py
# code object from '/usr/lib/python3.6/__pycache__/copyreg.cpython-36.pyc'
import 'copyreg' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c268e72b0>
import 're' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c26fa67f0>
# /usr/lib/python3.6/__pycache__/traceback.cpython-36.pyc matches /usr/lib/python3.6/traceback.py
# code object from '/usr/lib/python3.6/__pycache__/traceback.cpython-36.pyc'
# /usr/lib/python3.6/__pycache__/linecache.cpython-36.pyc matches /usr/lib/python3.6/linecache.py
# code object from '/usr/lib/python3.6/__pycache__/linecache.cpython-36.pyc'
# /usr/lib/python3.6/__pycache__/tokenize.cpython-36.pyc matches /usr/lib/python3.6/tokenize.py
# code object from '/usr/lib/python3.6/__pycache__/tokenize.cpython-36.pyc'
# /usr/lib/python3.6/__pycache__/token.cpython-36.pyc matches /usr/lib/python3.6/token.py
# code object from '/usr/lib/python3.6/__pycache__/token.cpython-36.pyc'
import 'token' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c26902278>
import 'tokenize' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c268f2978>
import 'linecache' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c268f2668>
import 'traceback' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c268e7588>
# /usr/lib/python3/dist-packages/apport/__pycache__/__init__.cpython-36.pyc matches /usr/lib/python3/dist-packages/apport/__init__.py
# code object from '/usr/lib/python3/dist-packages/apport/__pycache__/__init__.cpython-36.pyc'
import 'time' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3/dist-packages/apport/__pycache__/report.cpython-36.pyc matches /usr/lib/python3/dist-packages/apport/report.py
# code object from '/usr/lib/python3/dist-packages/apport/__pycache__/report.cpython-36.pyc'
# /usr/lib/python3.6/__pycache__/subprocess.cpython-36.pyc matches /usr/lib/python3.6/subprocess.py
# code object from '/usr/lib/python3.6/__pycache__/subprocess.cpython-36.pyc'
# /usr/lib/python3.6/__pycache__/signal.cpython-36.pyc matches /usr/lib/python3.6/signal.py
# code object from '/usr/lib/python3.6/__pycache__/signal.cpython-36.pyc'
import 'signal' # <_frozen_importlib_external.SourceFileLoader object at 0x7f7c268b0b38>
import '_posixsubprocess' # <class '_frozen_importlib.BuiltinImporter'>
import 'select' # <class '_frozen_importlib.BuiltinImporter'>
# /usr/lib/python3.6/__pycache__/selectors.cpython-36.pyc matches /usr/lib/python3.6/selectors.py
# code object from '/usr/lib/python3.6/__pycache__/selectors.cpython-36.pyc'
[... 292 lines cut ...]
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from python3...Reading symbols from /usr/lib/debug/.build-id/28/7763e881de67a59b31b452dd0161047f7c0135.debug...done.
done.
(gdb) run test.py
Starting program: /usr/bin/python3 test.py
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff7d362a9 in GlobalError::PushToStack() () from /usr/lib/x86_64-linux-gnu/libapt-pkg.so.5.0
```

Finally, I repeated this on a different machine also running Ubuntu 18.04 to the same result.

Perhaps long-term packing it up as pip module could help simplify this, or at least keep track of it. It's a great feature.

carrotIndustries, 2020-03-26:
Since you already have `gdb`, please provide a stacktrace by running `bt` in the gdb shell.

carrotIndustries, 2020-03-26:
The python module was missing some symbols. 22dde421b516228005ba54572b0f313c500c5854 fixed this. Don't know if that could cause a segfault (it didn't on my system)

iovaris, 2020-03-26:
Updating to 22dde421b516228005ba54572b0f313c500c5854 fixed the issue for me. Thanks!

closed 2020-03-26 by iovaris