horizon-eda/horizon #361: Copy schematic does not work properly - some of selection is not copied

Pitrsek, 2020-03-27:
Version used:horizon-2020-03-23-2339
[image]

All nets were selected(with pink rectangles around), ctrl+c, ctrl+v - bottom part of the picture is the result of paste. Not all net wires were copied

iovaris, 2020-03-27:
I have noticed this as well with buses. Bus rippers, bus labels, and wires connected to bus rippers don't seem to copy/paste correctly.

carrotIndustries, 2020-03-27:
Yes, there are some limitations in copy/paste since I didn't bother to implement copy/paste for all object types back then. I'll have a look.

carrotIndustries, 2020-03-27:
The net line copy/paste should be fixed by 41c5d1e117f017207c45ea802bdaff183fe69ce8

Pitrsek, 2020-03-28:
Might be related - when I select power symbol with rectangular selection, power symbol is highlighted(only power symbol, connected net is not selected), when I press delete, power symbol AND connected nets are deleted. If I select power symbol by clicking on it and press delete, it behaves correctly(net is not deleted)

carrotIndustries, 2020-03-28:
> Might be related - when I select power symbol with rectangular selection, power symbol is highlighted(only power symbol, connected net is not selected), when I press delete, power symbol AND [...]

This is unrelated and intended behaviour: When using the box select, you also select the junction the power symbol and net line are attached to. Deleting a junction causes all attached net lines to be deleted as well.

Pitrsek, 2020-03-28:
In that case I believe that the net should be highlighted with the pink rectangle as well. Currently there is no indication that nets are are selected the nets as well - so there is no expectation that the delete command would delete them too. 

Personally I do not expect some command to be executed on something that I have not selected(either explicitly with selection tool, or implicitly by hovering cursor over it). I'd select the nets with the selection tool if I wanted to delete them. Currently there is no difference weather I include nets in the selection, or I only select power ports. 

I'll try to think about how to include this feature in the workflow. It's a new concept to me. 
I'm aware that I can use selection filter to just select ports.

carrotIndustries, 2020-03-28:
> In that case I believe that the net should be highlighted with the pink rectangle as well. 

No, since moving would then move the other junction of the net line as well.

> Currently there is no indication

There is, albeit a rather minor one: The head-up display to the top right indicates that either just the power symbol or power symbol+junction are selected.

Don't expect this behaviour to change significantly in the near future. Selecting with the box select tool will select everything with the selection origin inside the box.

Anyhow, I'm closing this since the issue itself got fixed.

closed 2020-03-28 by carrotIndustries